jeelabs/esp-link #294: Integration in Arduino

pgollor, 2017-06-05:
Hi everyone.,
first of all i would to says this project is a great one. :-)

I have a question. Is it possible to flash an Arduino directly from the Arduino GUI?
In the GUI i get under ports the esp link (this information comes from mDNS). But if i try to flash it i get the following error:
```
/usr/local/arduino-1.8.2/hardware/tools/avr/bin/arduinoOTA -address XXX.XXX.XXX.XXX -port {upload.network.port} -sketch /tmp/arduino_build_970563/Ofentermometer.ino.hex -upload {upload.network.endpoint_upload} -sync {upload.network.endpoint_sync} -reset {upload.network.endpoint_reset} -sync_exp {upload.network.sync_return} 
Resetting the board
Failed to reset the board, upload failed
```
`XXX.XXX.XXX.XXX` is the right address, but the other parameters are not be available.

- It is possible to use the [ArduinoOTA](https://github.com/arduino/arduinoOTA) script from above?
- Or could you integrate the `avrflash` tool into the GUI like [this one](https://github.com/esp8266/arduino-esp8266fs-plugin) under the tools menu?

Thanks.

tve, 2017-06-05:
Thanks for your interest in esp-link! Yes, integration into the GUI is theoretically possible, but someone has to write the code for it... I never really used the GUI and am mostly using stm32 processors at this time, so I cannot spend the time to do the integration, sorry. Pull requests always welcome.

WRT the failure you're observing, I would highly recommend you use the avrflash script that comes with esp-link. It uses direct HTTP requests to esp-link to do the programming and will give you better error messages. Once that works, you can try arduinoOTA again (I didn't know that existed).

jlegen, 2017-06-05:
i am doing exactly this since quite a time using this great esp-link software - i am flashing Arduino devices installed elsewhere right from the Arduino IDE. All you need for this is a virtual com port (using a software like e.g. "netburner") pointing to the IP & port of the target...

pgollor, 2017-06-05:
Thanks for your posts. For me it is not problem to use the `avrflash` script or another solution. But I'm locking for a easy GUI solution for my project group in my university. An i have the same problem like @tve: the time ;)

@tve do you have any (easy to access) information for flashing an STM32 chip with esp-link? We are using STM32 boards too. Thanks.

jlegen, 2017-06-06:
not sure if i understood your problem - i also meant this to be an "easy GUI solution". I am using the Arduino IDE and click on "compile & upload" - that's it.
You only have to configure this *once* (for a target); with Linux, this is even more easy, and only a one-liner, e.g.:
socat  pty,link=/dev/virtualcom0,raw  tcp:192.168.254.254:8080&

tve, 2017-06-06:
> Thanks for your posts. For me it is not problem to use the avrflash script or another solution.

You need to make sure it works using avrflash first, one that works, use the virtual com port.

> do you have any (easy to access) information for flashing an STM32 chip with esp-link?

Dunno about "easy" :-). I use stm32L0 and stm32F1 series that can be reprogrammed over serial and I use the update command in https://github.com/jeelabs/folie but you would likely want to extract the update code and make your own tool.

pgollor, 2017-06-10:
thanks for your info @tve

closed 2017-06-10 by pgollor