jeelabs/esp-link #295: CLKOut

clawham, 2017-06-09:
Hello!
There is little problem - flashing AVRs w/o quartz that flashet to run from it before! In order to flash these processors - they need to send a clock signal of 1 MHz. Any PWM Gpio may be used!

tve, 2017-06-09:
I'm not sure I'm parsing your sentence correctly, but esp-link does not support outputting a clock on a gpio pin. The esp8266 has some very primitive PWM support and it may be possible to use it for this purpose but someone has to write the code for it ;-).

clawham, 2017-06-09:
I run win10x64 and can`t esplink fw from sources! Sorry-i`m beginner...
but there is sample code to run pwm!
`#include "driver/pwm.h"

#define PWM_0_OUT_IO_MUX PERIPHS_IO_MUX_MTDI_U
#define PWM_0_OUT_IO_NUM 12
#define PWM_0_OUT_IO_FUNC FUNC_GPIO12

#define PWM_1_OUT_IO_MUX PERIPHS_IO_MUX_MTDO_U
#define PWM_1_OUT_IO_NUM 15
#define PWM_1_OUT_IO_FUNC FUNC_GPIO15

#define PWM_2_OUT_IO_MUX PERIPHS_IO_MUX_MTCK_U
#define PWM_2_OUT_IO_NUM 13
#define PWM_2_OUT_IO_FUNC FUNC_GPIO13

uint32 io_info[][3] = {   {PWM_0_OUT_IO_MUX,PWM_0_OUT_IO_FUNC,PWM_0_OUT_IO_NUM},
       	                  {PWM_1_OUT_IO_MUX,PWM_1_OUT_IO_FUNC,PWM_1_OUT_IO_NUM},
               	          {PWM_2_OUT_IO_MUX,PWM_2_OUT_IO_FUNC,PWM_2_OUT_IO_NUM},
			};

u32 duty[3] = {512,512,512};

void ICACHE_FLASH_ATTR user_init(void)
{
	pwm_init(1000, duty, 3, io_info);
	pwm_start();
}

`
If someone can build the firmware with the addition of this code - check whether any frequency appears on the gpio 12 13 15. The frequency is generally not important.

tve, 2017-06-16:
I recommend you try this as a stand-alone program first...

clawham, 2017-06-16:
i`d tried and it works! This is an example in SDK! Bui шёь cannot compile your sources from win10x64. I`m not linux man %)

tve, 2017-06-17:
To compile on windows use the docker image...