jeelabs/esp-link #297: Debug problem in VisualMicro

Kabron287, 2017-06-09:
Hello! Great project.
I successfully forced it to work with VisualMicro, flashing Wemos D1 mini lite board up to 460800 baud directly from Visual Studio.
BUT, I could not debug programs. Entering debug session executed OK. Trouble begin with  Step/Continue operation. The program after launching debug session stops at the first breakpoint. When I execute a step to next breakpoint, VM debugger sends 0x63 which for unknown for me reason push down ISP/Flash Pin and step does not executed.
Is it possible to avoid such reaction?
Thanks in advance,
Vladimir

tve, 2017-06-09:
Nice set-up! On port 23 esp-link uses the telnet escape sequences, which all start with "IAC", which is character code 255 (0xff). So unless something else is going on, code 0x63 should not be a problem. You can try port 2323 and see whether that behaves differently.

Kabron287, 2017-06-09:
Thanks for answer, tve. 
I do use 2323 and Tibbo VSP, because 23 does not control ISP/Flash Pin.
I tested  ISP/Flash Pin behaviour by direct sending 0x63('c') from terminal. This code definitelly switch pin to ground. That is unacceptable. VisualMicro uses esptool for programming and then instantly switched to internal debugger.

tve, 2017-06-09:
> VisualMicro uses esptool for programming and then instantly switched to internal debugger.

This sounds like esptool does one connection, and then the debugger opens a fresh connection. If you're using port 2323 the debugger's connection will cause a reset. Maybe that's what you're seeing and you should use port 23 instead...

Kabron287, 2017-06-10:
Well, let me to sum up the situation:
1. The debugger's connection does not cause reset. It is absolutelly clear.
2. Port 23 is working, but unusable, because esptool does not control Reset and Flash pins.
3. When port 2323 used, flashing goes on perfectly, but during debug upon receiving byte 0x63 (and only in this case) your firmware made Reset and Flash pins low, causing the Target to fell into Flash mode. See attachment.

That is all.

If, after programming, I manually switch to port 23, I could execute single step successfully by sending byte 0x63. But it is impossible to make such switching on the fly inside the VisualMicro.

So, all I need is to disable actuating Reset and Flash pins in reaction of receiving 0x63 from port 2323.
Could it be realized?  

[image]

Kabron287, 2017-06-10:
Find working solution:
Menu>VMICRO>Debugger>Override Local Port
Thus I use VSP COM3:2323 for Flashing then debug automatically switch to COM4:23

closed 2017-06-10 by Kabron287

tve, 2017-06-10:
Great, I'm glad you found a work-around!

> So, all I need is to disable actuating Reset and Flash pins in reaction of receiving 0x63 from port 2323.
Could it be realized?

You're misunderstanding, it's not the 0x63 that triggers the reset, it's the first character on a connection to port 2323. This feature is what makes your esptool connection work.

Kabron287, 2017-06-10:
"it's the first character on a connection to port 2323. "
Definitelly NOT, because if I connect via any terminal program, nothing happened until I send 0x63 manually. And only 0x63 causes this. I scanned all bytes from 0x00 to 0xFF.
Why do not try it youself?