jeelabs/esp-link #298: Flash is good, but access point never comes up so I can configure device

spens, 2017-06-15:
Hi All, I've been working on this for days and I'm finally able to get the flash piece working on the first try for new ESP-01 modules.  Everything looks right, but the access point never comes up.  I've tried 4 different ESP-01 modules from 2 different sources.  I've had modules come online a while back (one ESP-01 and one ESP-12), but I can't get it to work now and I'm not sure what I'm doing wrong/differently.

Any suggestions on how to troubleshoot from this point?

Here is the output from my upload.  I don't need to configure any of these files right?  If I remember, I just flash to the device then connect to the access point to configure.

```
python "C:/MyData/Robotics/esptool/esptool.py" --chip esp8266 --port COM5 --baud 115200 write_flash -fs 1MB -ff 40m 0x00000 boot_v1.5.bin 0x1000 user1.bin 0xFC000 esp_init_data_default.bin 0xFE000 blank.bin

esptool.py v2.0-beta3
Connecting....
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash params set to 0x0020
Compressed 3232 bytes to 2336...
Wrote 3232 bytes (2336 compressed) at 0x00000000 in 0.2 seconds (effective 116.7 kbit/s)...
Hash of data verified.
Compressed 304884 bytes to 226120...
Wrote 304884 bytes (226120 compressed) at 0x00001000 in 20.0 seconds (effective 122.1 kbit/s)...
Hash of data verified.
Compressed 128 bytes to 75...
Wrote 128 bytes (75 compressed) at 0x000fc000 in 0.0 seconds (effective 71.3 kbit/s)...
Hash of data verified.
Compressed 4096 bytes to 26...
Wrote 4096 bytes (26 compressed) at 0x000fe000 in 0.0 seconds (effective 2208.8 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting...
Press any key to continue . . .

````

I've got batteries (18650s) that can source as much as 20A and I'm using a Buck Converter that is holding steady at 3.29 volts.

I've tried to find the network with my laptop and my phone and I don't see it.

Thanks!

uzi18, 2017-06-15:
What version did You flashed?

yesyesuk, 2017-06-15:
Are you sure the ESP module has a 8Mbit flash chip? I've seen it doing weird things (like not providing an AP) if the flash size is wrong when flashing, probably because it can't find its settings. 

My ESP-03 modules come with different flash sizes each time I order new ones. 

You could check the markings on the flash chip and google them to find out the size/capacity.

You need to adjust the addresses for the files depending on the flash size.

yesyesuk, 2017-06-15:
These are the locations I use for 4Mbit flash size
[image]

spens, 2017-06-15:
Thanks for all the replies folks!!  I really want to get this working so I can flash my Arduinos over WiFi.

From what I can tell, they are ESP-01 modules with 1MB.
First Amazon order: [https://www.amazon.com/gp/product/B016U051MK](https://www.amazon.com/gp/product/B016U051MK) 
Second Amazon order: [https://www.amazon.com/gp/product/B071HCX3X7](https://www.amazon.com/gp/product/B071HCX3X7)

Command line is in the original post, but here are the locations:
```
0x00000 boot_v1.5.bin 
0x1000 user1.bin 
0xFC000 esp_init_data_default.bin 
0xFE000 blank.bin

```

I've also tried boot_v1.7.bin but then reverted back when it wasn't working.

Bill

uzi18, 2017-06-15:
Try version 3.0alpha

swilson86, 2017-06-15:
I have been successfully using ESP-01 modules on eBay for about a year now and flashing with esp-link 2.2.3  Every one of them I bought has a 1MB flash (8 Mbit). This allows you to update the Arduino over the air. For that I use avr-flash at a command prompt on Wi
[image]
ndows.

spens, 2017-06-15:
I'll give these settings a try.  This is the second screen shot of the download tool.  When I use the download tool, it randomly fails when the progress bar is 10% or 50% through the download.  That is why I switched to esptool.  I've got two different USB serial adapters.  I'll try them again when I get home.

spens, 2017-06-17:
Eventually I was able to get the ESP8266 Download Tool to work successfully.  I did need to include the esp_init_data_default mentioned in an earlier post, but I used the addresses for a 1MB chip.

After 10-20 tries per device, I was able to get a successful download on 4 of the 5 devices.  However, only 1 device was showing up on my phone or computer.  It happen to be the same batch as a module I got working a couple of weeks ago.  None of my new batch of modules was showing up as an AP even though I was able to flash them successfully (and I think the tool I was using may have been flashing properly as well, it is just that the device doesn't seem to be working even though the red LED is on and the blue one blinks at startup).

I'll return the batch of 5 devices that I purchased and try some different ones.

I do have 2 questions maybe someone can answer.
1) After you flash, what is the minimum steps you need to do to see the AP?  Should it show up immediately, do you need to just cycle the power, or do you have to pull GPIO-00 high and cycle the power?  I don't have enough successes to know when I should see it so I just keep doing things until it shows up.

2) How does esp-link choose what channel to transmit on?  When I set up a typical access point for a home network, I have to pick what band to run on.  Isn't there something similar for these WiFi modules?  Strangely enough, even though my phone and computer picked up the device as an access point I could connect to, my android app that shows all the access points in my area and what frequency bands they are using does not pick up the ESP device.

Thanks again for everyone's help!
Bill

swilson86, 2017-06-17:
I have never flashed an ESP-01 using the esp_init_data_default.bin file. The other 3 files are the only ones I use. Here are some other things to check... CH_PD on the ESP-01 must be connected to 3.3V throughout the flashing and then when communicating. GPIO0 is jumpered to ground during flashing. Reset pin is momentarily shorted to ground just prior to flashing. I am using an FTDI232 chip and a USB cable connected to a laptop to flash the chip. After flashing unplug the USB cable, remove the jumper to ground on GPIO0 and then plug in the USB cable again. Since the USB port does not have a lot of power I have to be very close to the Windows laptop I am using to talk to the ESP-01 with WiFi. On my laptop I see the chip come up within 2 seconds of USB power on when I click on the Windows WiFi icon. I connect to the chip via WiFi and then open Chrome browser to 192.168.4.1 and voila! Now you can set all kinds of things in the esp-link software that you can visually see.

Let me know if you need more details on any section of the above paragraph.

uzi18, 2017-06-17:
You need special compilation for 1MB version.

spens, 2017-06-24:
OK, I got in a new set of chips.  Instead of being stuck into a piece of white Styrofoam, these were individually packaged like the first set I bought.  They are ESP-01s from here:
Amazon: https://www.amazon.com/gp/product/B01N98BTRH
[image]

Here is my setup for flashing the ESP-01s
[image]

I use batteries and a buck converter to get 3.3v because my FTDI cable doesn't seem to have enough juice to drive the ESP when it is connected.

Here it is with all of the proper connections:
[image]

When I upload, sometimes I get an error (could be at the beginning or often on the last step of the progress bar)
[image]

However after 1-3 tries, I usually get a successful upload:
[image]

Also, per the earlier comment, I don't need the esp_init_data_default.bin, but it didn't work the first time, so I added it to give it a try.  It still didn't work, but I didn't take it back out (the directions on this site say to add it and I can't find a good reference on google to say what that file is actually for)

This is where I still have trouble.  After successful upload, I removed the FTDI cable, replaced the IO0 to ground with a 1K resistor to 3.3v, put an LED across IO2.  I unplugged the power to the board, then plugged it back in.  I did not see any new access point on my computer, my phone, or even other devices I have (e.g. kindle).
[image]

I started changing things like tying the reset pin high, removing the resistor and connection IO0 directly to ground.  Connecting reset to ground then to +3.3.  Etc...  At one point, my device came alive and I was able to connect and configure it.  I never got the next two show up and I still haven't found a reliable pattern to get it to come up on the first try.

Any other ideas?

Bill

tve, 2017-06-27:
For a normal boot (not flashing) you need IO0 and IO2 with a pull-up.
If you're not seeing the access point, look at the output into your FTDI dongle, it should print boot messages at 115200 baud. If that has nothing connect IO2 to RX on your FTDI and see whether that says anything at boot, could be either 74880 baud or 115200 baud (it's "uart1").

yesyesuk, 2017-08-06:
I have the same (or similar) issue with one of my ESP-03 modules. It does not show up as an access point. I tried the instructions above (checked the serial output on boot). It prints this in a loop. Any idea? I have many more modules working fine, just this one doesn't want to work.

Fatal exception (0):
epc1=0x401000b8, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0):
epc1=0x401000b8, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0):
epc1=0x401000b8, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000