jeelabs/esp-link #299: MQTT Connect/Disconnect every few seconds

swilson86, 2017-06-17:
Configuration: I have approx 20 ESP-01 modules connected with Arduino's on the same network. I am running Mosquitto on a Raspberry Pi 3 as the Broker/Server. The ESP's are all running v 2.2.3 of esp-link and arduinos have el-client installed. I have the log file on each esp-link turned off to further save processing.

1) What is the cause of MQTT Disconnected/Connected messages on the uC Console every few seconds? In my Arduino loop I have the very first line always: 

esp.Process(); 

and I am suspicious this is part of the problem. I have tried: 

if (!connected) 
esp.Process(); 

with mixed results. I am also suspecting as my network grows the traffic may be the cause of the MQTT Disconnected/Connected messages.

2) I almost always publish retained messages to ensure that a subscriber catches the message due to the frequent MQTT disconnects. This seems to be a good solution as you don't support anything other than QOS 0. The bad side effect of this is that after every Disconnect I get all the messages again (every few seconds).

3) There is a lot of excess verbiage in esp-link on the uC Console... ie: "ELC: got" and then hex code., and same with "RESP_CB" also with hex code. I never use this info, and if I did it would be nice to be able to read it.

4) I would love to volunteer to create examples of code and documentation using "Best Practices." Your code is absolutely fabulous but documentation is the weakest area from my point of view. For almost a year now I have been fighting with esp-link / el-client and tested and retested.

tve, 2017-06-17:
0. I would recommend the 3.0.14.alpha image. It has a good number of bug fixes. I am going to promote it to release as soon as I get my act together and set-up my test-bench again.

1. Please paste the debug console output here so I can see what it says.

2. Makes sense.

3. I thought the newest version of el-client allowed you to compile with the debug stuff properly disabled.

4. Pull requests are welcome. There's actually quite a lot documented in the various markdown files for esp-link. If you're talking about the el-client portion then you're correct.

swilson86, 2017-06-17:
Thanks for the speedy response. I appreciate that you have been working 18 hours a day lately on getting 3 alpha ready and doing a ton of work! Here is the debug console output below. I will then flash for 3.0.14 and test again.

: 4=restart
   406> exccause=0 epc1=0x0 epc2=0x0 epc3=0x0 excvaddr=0x0 depc=0x0
   406> Flash map 1MB:512/512, manuf 0xE0 chip 0x4014
   406> ** esp-link v2.2.3 - 2016-06-21 21:58:48 - 1bcdc62: ready, heap=25112
   407> initializing MQTT
   407> MQTT_Init, host=192.168.0.40
   407> MQTT: Connect to 192.168.0.40:1883 0x3fff5e38 (client=0x3fff5574)
   407> MQTT ERROR: Failed to connect
   408> initializing user application
   408> Waiting for work to do...
   408> mode : sta(18:fe:34:d4:b3:22) + softAP(1a:fe:34:d4:b3:22)
   408> add if0
   409> add if1
   409> dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
   410> bcn 100
   411> f r0, scandone
  2532> state: 0 -> 2 (b0)
  2534> state: 2 -> 3 (0)
  2537> state: 3 -> 5 (10)
  2537> add 0
  2537> aid 5
  2537> cnt 
  2550> 
  2550> connected with Buffalo2, channel 6
  2551> ip:192.168.0.28,mask:255.255.255.0,gw:192.168.0.1
  2551> Wifi connected to ssid Buffalo2, ch 6
  2551> Wifi got ip:192.168.0.28,mask:255.255.255.0,gw:192.168.0.1
  2552> MQTT: Connect to 192.168.0.40:1883 0x3fff7198 (client=0x3fff5574)
  2598> MQTT: TCP connected to 192.168.0.40:1883
  2599> MQTT: Send type=TYPE_CONNECT id=0000 len=43
  2609> MQTT: Recv type=CONNACK id=0000 len=4; Pend type=NULL id=00
  2920> MQTT: Disconnect CB, freeing espconn 0x3fff7198
  2920> MQTT: Disconnected from 192.168.0.40:1883

tve, 2017-06-17:
```
2552> MQTT: Connect to 192.168.0.40:1883 0x3fff7198 (client=0x3fff5574)
2598> MQTT: TCP connected to 192.168.0.40:1883
2599> MQTT: Send type=TYPE_CONNECT id=0000 len=43
2609> MQTT: Recv type=CONNACK id=0000 len=4; Pend type=NULL id=00
2920> MQTT: Disconnect CB, freeing espconn 0x3fff7198
2920> MQTT: Disconnected from 192.168.0.40:1883
```
The above says that the TCP connection to the MQTT server got established and 311ms later a disconnect was received. I wonder whether there's an authentication problem, or perhaps the server requires SSL, or? You should look into your MQTT server logs and see what it says (may have to turn some debug logging on).

swilson86, 2017-06-18:
Duh! You were right... two publishers had the same name and this caused MQTT to continually disconnect/reconnect the two stations. Thanks!

closed 2017-06-18 by swilson86