jeelabs/esp-link #300: Many MQTT subscriptions don't work

tve, 2017-06-18:
From gitter https://gitter.im/jeelabs/esp-link?at=5935847b31f589c64f787f78 :

> Hi! I tried 3.0.14 Alpha some time ago (maybe 2 month) and found a bug: I didn't received mqtt messages from some of subscriptions (I have about 10 in my project), I also tried the latest el-cleint library with same result. So I had to revert to 2.2.3 (wich I'am using about an year)  Was this issue fixed in the time past from 3.0.14 Alpha release?

> I am using arduino & el-client to make subscriptions. I found that arduino doesn't receive messages from the last subscriptions. I also found strange lines (mqtt related) in esp-link log, but I don't remember what exactly, sorry. I went back to 'stable' release and problems are gone. So, if it is first time when you hear it... just check how elclient&esp-link works with some amount of subsriptipn (10-20, each topic 15-25 length)... Also I found that el-client still contains some serial debug output even if debug serial isn't configured in arduino sketch. Thank you for your work...

tve, 2017-06-18:
Looks like a string of mqtt.subscribe calls just overrun the uart buffer. I believe the best solution would be to add an explicit ACK.

tve, 2017-06-27:
the work-around works for me, not ideal, but not that horrible either.

Familienpapi, 2017-09-02:
this problem exists also in pubsubclient. 
The workaround there is to call the mqttclient.loop() command after every subscription. 
I have only two subscriptions but I do the esp.Process() after every subscription. Maybe this helps here, too.