jeelabs/esp-link #302: Firmware edit to post data to domoticz server

delchrys, 2017-06-21:
Hi all.
I was wondering if I can make a change to the firmware so the esp8266 sends decoded serial data to a domoticz server.
It had to update a value in a database like this:
http://192.168.1.2:8080/json.htm?type=command&param=udevice&idx=$idx&nvalue=0&svalue=79
Just a simple post or get command. Is this possible by editing some things in the current Firmware?
Connecting a arduino is also possible but I think the esp8266 can do this simple task on it's own.
Please some advice or pointers where to go.

tve, 2017-06-21:
All the serial input comes through https://github.com/jeelabs/esp-link/blob/master/serial/serbridge.c#L465-L479 where it gets dispatched to the uC programming code, or std "console" code, or SLIp interpretation. You would hook in there and dispatch to your own code that would parse line by line (?) your data and then make a rest request. The request code is around https://github.com/jeelabs/esp-link/blob/master/rest/rest.c#L343

delchrys, 2017-06-21:
So if I edit there I then can rebuild the firmware files and reflash? Also need a alternated uart to decode the serial data. Something like :
https://github.com/susisstrolch/ems-esp-link?files=1
He made some changes so it works with a certain protocol I also need