jeelabs/esp-link #305: Question

JacoFourie, 2017-06-27:
Hi all. Please forgive me if this question has been asked. I am using the Arduino Uno WiFi Dev boards. The Firmware on these boards seem to be very buggy and I dont get any answers over at the Arduino Forum about it. It will keep on switching from STA mode to AP+STA mode. When this happens the MQTT client will disconnect and reconnect from the broker all the time.  Only a wifi reboot and changing the mode back to STA will fix this problem. It will also lose some of the topics it subscribes to. This happens randomly. Only a reboot of the micro controller will fix that. So my question is will this firmware fix those issues and can it run on the Uno board ?

Kind Regards.

tve, 2017-06-27:
To answer your last question first: yes, it can run on the uno. There are others that have done it. I don't have an uno so I can't provide concrete help.

I would consider random switching between sta and sta+ap a bug. I have not seen that. I'm actually also wondering why it would loose the MQTT connection in that process, so I'm actually wondering whether something else may be going on. For example, it may loose the connection to the AP due to signal strength. There is code in esp-link to turn the AP back on again in that case but I'm pretty sure I disabled that (but arduino.cc may have kept it in). But the main point is that you may be loosing your MQTT connections due to getting disconnected from the AP, not because of a switch to STA+AP mode.

JacoFourie, 2017-06-27:
Hi. Thanks for the reply. I am not the only one getting this issue. If you have a look at this post http://www.arduino.org/forums/uno/arduino-uno-wifi-keeps-reconnecting-to-mqtt-server-1726.
The publish keeps on working. I have a sketch that will publish values every 3 seconds and that keeps ons working. But the subscribe stops so I assume it must be a bug. I saw some other people also 

How do I load this firmware on the Uno Wifi as the firmware I loaded is selected from a dropdown in the Arduino IDE.