jeelabs/esp-link #306: ESP-01 AP webserver not woeking

pwrampde, 2017-06-27:
Hi, I have a problem with ESP-01 - flashing is finished fine, I can see A, but I can nnot connect to ESP-01 as DHCP client - laptot does not receive IP and also signal strenght is low too. I tried also to delete ESP-01 and reflash it, but without success. Any idea what next?
I have also ESP-12 on WeMos D1 mini, which works fine without any problems, but ESP-01 looks more usable for my project.

Here is debug info from ESP-01.

 112> ** esp-link v3.0.14-g963ffbb
   113> Flash config restore *FAILED*
   113> CONN led=0
   113> SER led=14
   115> Wifi init, mode=AP
   117> Wifi uses DHCP, hostname=esp-link
   121> "ip": "0.0.0.0"
   123> "netmask": "0.0.0.0"
   126> "gateway": "0.0.0.0"
   128> "hostname": "<null>"
   131> sleep enable,type: 2
   134> Httpd init, conn=0x3fff3820
   137> No user file system found!
   140> Serbridge pins: reset=12 isp=13 swap=0
   153> Reset cause: 6=external
   153> exccause=0 epc1=0x0 epc2=0x0 epc3=0x0 excvaddr=0x0 depc=0x0
   153> Flash map 512KB:256/256, manuf 0xC8 chip 0x4013
   158> ** esp-link v3.0.14-g963ffbb: ready, heap=21368
   163> SNTP timesource set to us.pool.ntp.org with offset 0
   168> initializing user application
   172> Waiting for work to do...
   175> mode : softAP(1a:fe:34:a3:9d:12)
   178> add if1
   180> dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
   188> bcn 100

tve, 2017-06-27:
> `180> dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)`

The above line says that the DHCP server is starting fine, so you should be able to connect to the AP and see it at http://192.168.4.1
If that is not working, then perhaps you're connceting to the wrong AP?
WRT signal strength: the esp-01 RF section is no good, I see about 9dB better reception on an esp-12. For this reason alone I don't use any -01's anymore.

pwrampde, 2017-06-29:
Yep, everything looks fine, but I have no response from ESP-01 AP when client tries to connect and on the client side the connection time out appears. I also tried monitor client side wiht wireshark and I have no response packet from ESP-01. When I used ESP-01 as a wifi client with Arduino, I did not focused any troubles.