jeelabs/esp-link #307: Pure C MQTT Client

Zageek1, 2017-06-30:
I am wanting to implement an MQTT client in pure C on an AVR not using an Arduino. This is because my current application is in pure C. Is there any option available?

Zageek1, 2017-07-11:
I have read through the source and figured out the structure of the SLIP communications. So far I have SLIP working via serial perfectly and I am communicating with ESP-Link and I am working on publishing a topic however its going to be easire to convert this from one single function to a library with helper functions but  that is going to take too long for my project.

 What I am instead doing is recreating my plain C project in C++ and I will modify the ELclient here to strip out the Arduino stuff as I see the EL Client elegantly takes care of some of the conditions. Why reinvent the wheel...

tve, 2017-07-27:
OK, let us know if you run into issues. Apologies for the lack of docs on this part.