jeelabs/esp-link #308: Arduino 101: Invalid CRC

joseamirandavelez, 2017-07-03:
I am using an Arduino 101. I am having the issue where the ESP8266 ROM outputs a bot message on UTXD and is causing the 101 to not boot (at least that is what it looks like). I followed the instructions in the documentation (below)

```
At boot time the esp8266 ROM outputs a boot message on UTXD, this can cause problems to the attached microcontroller. If you need to avoid this, you can configure esp-link to swap the uart pins. You should then connect the esp-12 module as follows and choose the "swap_uart" pin assignment in the esp-link web interface:

GPIO13: connect to TX of microcontroller
GPIO15: connect to RX of microcontroller and use a pull-down to ensure proper booting
GPIO12: connect to RESET of microcontroller
GPIO14: connect to ISP of LPC/ARM microcontroller (not used with Arduino/AVR)
GPIO0: either a 1k-10k pull-up resistor to 3.3v or a green "conn" LED via a 1k-2.2k resistor to 3.3V (indicates wifi status)
GPIO2: either a 1k-10k pull-up resistor to 3.3v or a yellow "ser" LED via a 1k-2.2k resistor to 3.3V (indicates serial activity)

The GPIO pin assignments can be changed dynamically in the web UI and are saved in flash.
```
[https://github.com/jeelabs/esp-link/blob/master/FLASHING.md](https://github.com/jeelabs/esp-link/blob/master/FLASHING.md)

and now the 101 works, but I am getting the following error:

```
EL-Client starting!
ELC: got 128 @A800EE1C: 2020 203E3635 3336 53 4C 49 50 3A 20 73 74 61 72 74 20 6F 72 20 65 6E 64 20 6C 65 6E 3D 30 20 69 6E 70 6B 74 3D 30 D A 20 20 36 33 35 36 3E 20 53 4C 49 50 3A 20 73 74 61 72 74 20 6F 72 20 65 6E 64 20 6C 65 6E 3D 30 20 69 6E 70 6B 74 3D 31 D A 20 20 36 33 35 36 3E 20 53 4C 49 50 3A 20 73 74 61 72 74 20 6F 72 20 65 6E 64 20 6C 65 6E 3D 31 30 20 69 6E 70 6B 74 3D 31 D A 20
ELC: Invalid CRC
CMD??
ELC: got 51 @A800EE1C: 2020 203E3238 3436 63 6D 64 52 65 73 70 6F 6E 73 65 3A 20 63 6D 64 3D 32 20 76 61 6C 3D 2D 31 34 37 36 33 33 34 30 38 38 20 61 72 67 63 3D 30 D A

```

I can't find a solution for this... Has anybody seen this? I am desperate here... I have been trying to make this work for three weeks...

joseamirandavelez, 2017-07-15:
I am asking this in the right place?

tve, 2017-07-27:
Sorry, I was out of order... Which versions of esp-link and el-client are you using?

joseamirandavelez, 2017-07-27:
I was using the latest versions as of a week ago... I don't think it get too many updates.

tve, 2017-07-27:
Please turn debug output on uart0 off in esp-link. It's getting mixed into the SLIP packets (shouldn't, but...). The line `ELC: got 128 @A800EE1C: 2020 203E3635 3336` implies that the packet length is 0x2020, which is bogus and the cmd code is 0x3336, which is also bogus.

Ola-Palm, 2018-05-14:
I'm having the same issue on a arduino uno. 
When i send mqtt message with size over 46 i get CRC error. 

ELC: got 62 @557: 3 49C 2 B 0 2F 65 73 70 2D 6C 69 6E 6B 2F 31 0 0 0 25 0 7B 22 77 65 66 73 73 64 66 74 2D 32 32 67 65 66 72 67 68 67 22 3A 22 73 61 73 73 73 64 73 64 66 74 22 A 7D
ELC: Invalid CRC
ELC: got 62 @557: 3 49C 2 B 0 2F 65 73 70 2D 6C 69 6E 6B 2F 31 0 0 0 25 0 7B 22 77 65 66 73 73 64 66 74 2D 32 32 67 65 66 72 67 68 67 22 3A 22 73 61 73 73 73 64 73 64 66 74 22 A 7D
ELC: Invalid CRC
ELC: got 62 @557: 3 49C 2 B 0 2F 65 73 70 2D 6C 69 6E 6B 2F 31 0 0 0 24 0 7B 22 77 66 73 73 64 66 74 2D 32 32 67 65 66 72 67 68 67 22 3A 22 73 61 73 73 73 64 73 64 66 74 22 A 7D A
ELC: Invalid CRC
ELC: got 62 @557: 3 49C 2 B 0 2F 65 73 70 2D 6C 69 6E 6B 2F 31 0 0 0 23 0 7B 22 77 66 73 64 66 74 2D 32 32 67 65 66 72 67 68 67 22 3A 22 73 61 73 73 73 64 73 64 66 74 22 A 7D A 0
ELC: Invalid CRC
ELC: got 62 @557: 3 49C 2 B 0 2F 65 73 70 2D 6C 69 6E 6B 2F 31 0 0 0 22 0 7B 22 77 73 64 66 74 2D 32 32 67 65 66 72 67 68 67 22 3A 22 73 61 73 73 73 64 73 64 66 74 22 A 7D A CF 2B
RESP_CB: 1180 2
Received: topic=/esp-link/1
data={"wsdft-22gefrghg":"sasssdsdft"
}
when size is 46 or below message arrives as expected, mostly.

ESP-01
esplink esp-link v3.2.47-g9c6530d
UART debug log: off