jeelabs/esp-link #311: dev boards w/esp-12*

treii28, 2017-07-07:
I just started playing around with this on an ESP-01 and an ESP-03 successfully. I want to try dropping it on one of my free 12f's this weekend. But I also have some dev-board type esp-12 modules such as the NodeMCU and the Feather Huzzah. I tried the esptool.py to program these with no success. Is there a trick to getting the firmware onto one of these boards?

tve, 2017-07-27:
Did you follow the instructions at https://github.com/jeelabs/esp-link/blob/master/FLASHING.md ?

hponso, 2017-09-12:
Hi, I've been trying to make this work with a nodemcu and a wemos D1mini for about four hours now and I can't get it done.
At first I just ran thee esptool.py command from  https://github.com/jeelabs/esp-link/blob/master/FLASHING.md for the 4Mbyte module and it flashed with no errors, this is the output.

`
~/dev/esp-link-v3.0.14-g963ffbb$ esptool.py --port /dev/ttyUSB0 --baud 230400 write_flash -fs 32m -ff 80m 0x00000 boot_v1.6.bin 0x1000 user1.bin 0x3FC000 esp_init_data_default.bin 0x3FE000 blank.bin
WARNING: Flash size arguments in megabits like '32m' are deprecated.
Please use the equivalent size '4MB'.
Megabit arguments may be removed in a future release.
esptool.py v2.1
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 230400
Changed.
Configuring flash size...
Flash params set to 0x004f
Compressed 3856 bytes to 2763...
Wrote 3856 bytes (2763 compressed) at 0x00000000 in 0.1 seconds (effective 247.2 kbit/s)...
Hash of data verified.
Compressed 330756 bytes to 244619...
Wrote 330756 bytes (244619 compressed) at 0x00001000 in 11.1 seconds (effective 237.9 kbit/s)...
Hash of data verified.
Compressed 128 bytes to 75...
Wrote 128 bytes (75 compressed) at 0x003fc000 in 0.0 seconds (effective 177.0 kbit/s)...
Hash of data verified.
Compressed 4096 bytes to 26...
Wrote 4096 bytes (26 compressed) at 0x003fe000 in 0.0 seconds (effective 8770.8 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting...
`

but none of the boards does nothing afterwards.
Then connected GPIO0 to ground, GPIO2 to pull-up and GPIO15 to pull-down in the nodemcu as the tutorial said, and got the same flashing result.
But the board still does nothing when GPIOs are connected in normal operation mode (I don't have a reset pin on my board to connect to GPIO0, its a FLIP32 autopilot).

I tried to read something in serial port using gtktem on ttyUSB0 but a could find what value of bps to use. Something shows when I push reset but  it's gibberish.

I tried with versions 2.2.3 and 3.0.14.

Can someone see where did I go wrong?
Thanks a lot in advance!!

maxtox, 2017-12-17:
i have nearly the same result with dev-board v1.

I can flash bin files and can start it  BUT

after i connect TX and RX and START Dev board again, it is death!!!! = NO WIFI, SERIAL SHOWS bootmode (3;6), but nothing happenes.

I have to REWRITE all RAM with BLANK.bin file and afterwards flash esp-link again, then i can start again to see wifi etc.. BUT if i connect RX and TX same procedure!!!!
P.S.: i used connection wit a level shifter and without...same result)

What do we wrong?

maxtox, 2017-12-17:
now tried the same with esp-12f

serial showing something funy:

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 2408, room 16 
tail 8
chksum 0xe5
load 0x3ffe8000, len 776, room 0 
tail 8
chksum 0x84
load 0x3ffe8310, len 632, room 0 
tail 8
chksum 0xd8
csum 0xd8

2nd boot version : 1.6
  SPI Speed      : 40MHz
  SPI Mode       : DIO
  SPI Flash Size & Map: 32Mbit(512KB+512KB)
jump to run user1 @ 1000

⸮⸮⸮⸮n⸮r⸮⸮n|⸮l⸮l`⸮⸮r⸮l⸮l⸮l`⸮⸮r⸮l⸮l⸮l`⸮⸮