jeelabs/esp-link #312: 'Client' mode

varvas, 2017-07-09:
Are there any plans for developing a "client mode" as well? So that I could connect one ESP board to one device and another board to the second one and pair them together (and use Wifi as "cable" between them).

hexagon5un, 2017-07-19:
If I understand this right, I'd like this too: a TCP client mode.

Use case: my oscilloscope hooked up to the network exposes a TCP server, and I would like to control it with the microcontroller attached to an ESP8266 to make a remote control panel and automation hub.

I have a current workaround: using netcat on my laptop to initiate both TCP client transactions and pipe them to each other (`ncat $esp-ip 23 -c 'ncat $scope-ip 5025'`)  but it would be great to cut out the middle-man.

Combined with SLIP mode, it would make a very nice microcontroller-to-ethernet bridge (as long as it's on the WiFi network).

uzi18, 2017-07-20:
Can't You connect with slip from microcontroller to another esp-link port
23?

hexagon5un, 2017-07-21:
@uzi18   I can't.  But maybe I'm overlooking something -- I only briefly went through the code looking for a TCP client.  It shouldn't be all that hard to do, so if it's already implemented and we just don't see it, that would be great.  

Maybe I'm just overlooking something.  If you know how, please show us!  :)

martinl88, 2017-07-21:
I've seen Chinese "wifi-to-serial" boxes out there that have the ability to be a client, so it shouldn't be impossible to implement.

My own use case is a link between a remote POS printer and a PC. Currently I'm using HW VSP3 to make the connection. But as the PC has proper serial ports already on-board, I'd like to cut out the middle-man as well.

uzi18, 2017-07-21:
Try simply connect with tcp_client example to another esp-link,  port 23.

hexagon5un, 2017-07-21:
Aha! Thanks @uzi18 for the reminder about the useful example code in a totally different repository: https://github.com/jeelabs/el-client/tree/master/ELClient/examples

I actually remember going down this path before, when I was writing my own MQTT stuff for the esp-link.  The code in ELClient was a good foothold, even though I wasn't using Arduino in my project. 

Anyway, it looks like I should be able to hack together what I need here.  Thanks for the pointer.  

And the OP can probably do what he/she wants too: just set up a server on one with an unused port, and then connect to it with the other.

closed 2017-07-27 by tve

martinl88, 2017-07-27:
Sad to see that this idea is not even worth considering :(
Oh, well..

tve, 2017-07-27:
It's not that it's not worth considering, it's that the way esp-link works is that someone cares enough to implement a new feature and contributes it in a pull-request. This particular one has come up before but nobody has stepped-up. Besides this, I'm actually contemplating ripping features out in order to make space for HTTPS...

martinl88, 2017-07-27:
Thanks for the clarification. Sounds reasonable.

uzi18, 2017-07-28:
I have client mode in my future plans,  stay tuned.