jeelabs/esp-link #313: Mega2560 flash over virtual serial

Badiboy, 2017-07-10:
**Devices**
- ESP-01
- Arduino Mega2560

**Firmware/Soft**
- esp-link v3.2.47-g9c6530d
- Windows 10
- Arduino IDE 1.8.3
- Tibbo Device Server Toolkit

**Problem**
I made a esp-link bridge to the Arduino Mega2560 (looks like original, but seems to be Chineese). Serial works fine in both directions. Reset from esp-link console works fine. Flashing fails with timeout (full log at the end):
>          avrdude: stk500v2_ReceiveMessage(): timeout
**If I press the Mega Reset button in time (like for boards withoud DTR) flashing works well.**

I saw https://github.com/jeelabs/esp-link/issues/237 and https://github.com/jeelabs/esp-link/issues/87. I'd upload latest 3.2 release where Mega2560 support was announced. But still have the issue. Tested on two more or less simillar Mega2560 boards.

**PS**
[Here](http://samopal.pro/arduino-esp8266/) in comments ones said that the reason is:
"STK500v2 in MEGA works only at 115200. While avrdude is pulling DTR, uploader is getting tired to wait for flashing command".

Full log:
>          Sketch uses 2848 bytes (1%) of program storage space. Maximum is 253952 bytes.
>          Global variables use 198 bytes (2%) of dynamic memory, leaving 7994 bytes for local variables. Maximum is 8192 bytes.
>          C:\Users\Alex\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino9/bin/avrdude -CC:\Users\Alex\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino9/etc/avrdude.conf -v -patmega2560 -cwiring -PCOM7 -b115200 -D -Uflash:w:C:\Temp\arduino_build_985981/BlinkTest.ino.hex:i 
>          
>          avrdude: Version 6.3, compiled on Jan 17 2017 at 12:00:53
>          Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
>          Copyright (c) 2007-2014 Joerg Wunsch
> 
>          System wide configuration file is "C:\Users\Alex\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino9/etc/avrdude.conf"
> 
>          Using Port                    : COM7
>          Using Programmer              : wiring
>          Overriding Baud Rate          : 115200
>          avrdude: stk500v2_ReceiveMessage(): timeout
>          avrdude: stk500v2_ReceiveMessage(): timeout
>          avrdude: stk500v2_ReceiveMessage(): timeout
>          avrdude: stk500v2_ReceiveMessage(): timeout
>          avrdude: stk500v2_ReceiveMessage(): timeout
>          avrdude: stk500v2_ReceiveMessage(): timeout
>          avrdude: stk500v2_getsync(): timeout communicating with programmer
>          
>          avrdude done.  Thank you.
>          
>          An error occurred while uploading the sketch

helitron, 2017-07-11:
Hi Badiboy, have you updated the bootloader on the Megas already ? Have had the same problem with my Arduino mega 2560 and after updating the bootloader problem was gone.

Best regards,
Erwin

Badiboy, 2017-07-11:
Hi @helitron. No, I tried with bootloader placed there from their birth.
Today I bought USBasp and updated the bootloader to the one from Arduino IDE (Tools / Burn bootloader). From the log I see it flashed this one:
> -Uflash:w:C:\Users\Alex\AppData\Local\Arduino15\packages\arduino\hardware\ avr\1.6.19/bootloaders/stk500v2/stk500boot_v2_mega2560.hex:i
The esp-link result is **totally the same**. No auto-flash, only with manual reset button push.

uzi18, 2017-07-12:
Please update bootloader and use dedicated megaflash script to upload
firmware.

Badiboy, 2017-07-12:
@uzi18 
1) I'm under Windows, so have problems with bash scripts. 
2) I want to program and flash boards from Arduino IDE: one button compilation and flashing is a very usefull feature. :) Is it possible to attach external loader (like megaflash) to the IDE toolchain?

uzi18, 2017-07-12:
So it is impossible now, as esp-link does support optiboot bootloader only
as megaflash standalone flasher.

You can rewrite script for windows.

helitron, 2017-07-12:
I'm using megaflash.sh under Windows 10, no problem with bash ! Look here:

https://www.howtogeek.com/249966/how-to-install-and-use-the-linux-bash-shell-on-windows-10/

My commandline for flashing a mega 2560 is

megaflash.sh 192.168.0.xxx mySketch.hex

Works great.

But I'm also want to program the megas directly out of the Arduino IDE, with the help of a custom programmer in the Arduino file programmers.txt. Tried something but had no success so far unfortunately.

uzi18, 2017-07-12:
> the-linux-bash-shell-on-windows-10/

it is possible to flash from IDE, but You need to patch platform.txt and
programmers.txt

helitron, 2017-07-12:
Thanks uzi18 , have you an idea about such patches ?

I tried this patch in programmers.txt: 

esplink3FCtrl.name=ESPLink3
esplink3FCtrl.communication=serial
esplink3FCtrl.protocol=stk500v2
esplink3FCtrl.program.protocol=stk500v2
esplink3FCtrl.program.tool=avrdude
esplink3FCtrl.program.extra_params=-Pnet:192.168.0.67:23

A new programmer "ESPLink3" popped up in the programmers list of the IDE but when I used this programmer a timeout occured unfortunately.

As mentioned already with the script "megaflash.sh" flashing is working without any problems.

uzi18, 2017-07-12:
Try port 2323

helitron, 2017-07-12:
Thanks for the hint but tried it already.

Badiboy, 2017-07-16:
It's a bit said that native avrdude still does not work with esp-link for Mega. 

Thanks all for answers. I had succeess with attaching the megaflash script to the Arduino IDE (in Win10), so you can use it (if anyone need it). I had no time to do it in "feng shui" style, so if anyone will do it better - I will appreciate you. :)

So. 
1) You need to set up ["Bash on Ubuntu"](https://msdn.microsoft.com/en-us/commandline/wsl/about). May be sh.exe from Git will also work.
2) Write the batch file like this:
>  set search1=C:
>  set replace1=/mnt/c
>  
>  set search2=\
>  set replace2=/
>  
>  set search3=/"
>  set replace3=
>  
>  set hex=%1
>  
>  setlocal enabledelayedexpansion
>     set "hex=!hex:%search1%=%replace1%!"
>     set "hex=!hex:%search2%=%replace2%!"
>     set "hex=!hex:%search3%=%replace3%!"
>     C:\Windows\Sysnative\bash.exe -c "/mnt/d/Arduino/ESP8266/megaflash.sh -v 192.168.4.1 !hex!"
>  endlocal

where
- search1/replace1 is a drive letter for temporary folder where Arduino IDE store hex-files (for me it's C:\Temp\xxx). 
- /mnt/d/Arduino/ESP8266/megaflash.sh is a path to megaflash script, where /mnt/d is a letter of your drive. For me it's D:/Arduino/....
- 192.168.4.1 is the IP of your Esp-link.

_... Will be ontinued in next comment._

Badiboy, 2017-07-16:
3) Find the platform.txt in the **User** folder. For me it's 
> C:\Documents and Settings\Alex\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.19\platform.txt 

4) Add to the end code like this:
> tools.esplink3.path=D:/Arduino/ESP8266
> tools.esplink3.cmd.path={path}/megaflash.bat 
> tools.esplink3.config.path=
> tools.esplink3.network_cmd=
> tools.esplink3.upload.params.verbose=
> tools.esplink3.upload.params.quiet=
> tools.esplink3.upload.verify=
> tools.esplink3.upload.params.noverify=
> tools.esplink3.upload.pattern="{cmd.path}" ""{build.path}/{build.project_name}.hex""

May be empty lines are not needed, had no time for experiments. Where:
- tools.esplink3.path=D:/Arduino/ESP8266 is a path to the batch file. May be also not needed.
- tools.esplink3.cmd.path={path}/megaflash.bat is full path and name of the batch file written above.

5) Edit boards.txt in the same folder. Find your board. 
For me it's:
> mega.name=Arduino/Genuino Mega or Mega 2560

6) Replace
> mega.upload.tool=avrdude

with

> mega.upload.tool=esplink3

That's all, folks. :) Now you can upload sketches from Arduino IDE.

helitron, 2017-07-16:
Thanks a lot for this, will try it out soon.

uzi18, 2017-07-30:
You will more like instead of point 6, modify programmers.txt
and add at the end:
`
esplink.name=EspLink-192.168.2.6
esplink.communication=usb
esplink.protocol=esplink3
esplink.program.protocol=esplink3
esplink.program.tool=esplink3
esplink.program.extra_params=192.168.2.6

esplink1.name=EspLink-192.168.2.7
esplink1.communication=usb
esplink1.protocol=esplink3
esplink1.program.protocol=esplink3
esplink1.program.tool=esplink3
esplink1.program.extra_params=192.168.2.7
`

but also need to change lines in point 4 from:
`
tools.esplink3.upload.pattern="{cmd.path}" ""{build.path}/{build.project_name}.hex""
`
into:
`
tools.esplink3.upload.params.verbose=-v
tools.esplink3.upload.pattern="{cmd.path}" "{program.extra_params}" "{build.path}/{build.project_name}.hex"
tools.esplink3.program.params.verbose=-v
tools.esplink3.program.pattern="{cmd.path}" "{program.extra_params}" "{build.path}/{build.project_name}.hex"
`

Now You can use more than one esp-links from list and use upload using programmer.

profisir, 2017-10-30:
Hello :)
I use ESP-01 + flash updated to 4M +  mega2560 rev3 in my project LED light.
Test serial comunication baud 115200 - succes from webGui espLink.
Arduino 1.8.3 - bootloader flash - megaload not in sync.
Try optibot for mega2560 - no sucess..
Today i try  new Arduino Beta 1.9 beta 22 Software...
Megaload  script programming !!

uzi18, 2017-11-08:
Megaload or megaflash?

profisir, 2017-11-10:
TEN Z PACZKI ESPLINK

uzi18, 2018-01-06:
@profisir have compared bootloaders and looks like it is almost the same.
Did You realy reflashed Arduino 1.8 version and it didn't work?
please put here hex from 1.8 and grom 1.9 I will try it.

profisir, 2018-01-06:
@uzi did not compare,
I use Arduino 1.8.3 daily for programming, my favorite and modified under esp-link avrflash and megaflash. bootloader from it via usbasp did not work with mega flash (out of sync) I tried with all MEGA what I have at home, all the same model.

somewhere I read to try the bootloader from the new arduino ...
I downloaded Arduino PR BETA1.9 build 22 then I uploaded the bootloader via usbasp from this Arduino  and it's works, updated the bootloaders in this way all my MEGA 2560 boards works now.

In another project (Car Heater Eberspacher D4WSC) where I have ArduinoProMicro 8Mhz + esp-01 (32Mbit Flash MOD) I had to add voltage converter from 3.3V esp to 5V ardu on two transistors + 4 resistors because ESP did not upload software - i dont remember error - but the communication in the console looked like worked ok , the program responded to queries but not programming arduino.

Now I will add a voltage converter for all new projects from ESP.

uzi18, 2018-01-06:
Maybe Your bootloader was very old,  it is possible in china clones.
Sometimes need to just reflash bootloader in new processor.

uzi18, 2018-01-06:
For converter part,  it usually enough to add rectifier diode in directiin
from esp tx to arduino rx

profisir, 2018-01-08:
@uzi i wrote : 
- on arduino 183 botloader ar not sync, on Beta 22 ar OK.
and esp8266 is 5V tolerant on Digital I/O, on RX esp no need diode, just simple resistor 1K, on TX esp in most cases 1K is OK but sometimes like my arduino pro mini + long cable u must add  LevelUP to 5V conv.

uzi, 2018-01-08:
@profisir I think you mean to @uzi18 and not me ... this way they get notified instead of me. 😛

frizzle101101, 2018-02-17:
uploading my bootloader started making the megaflash script work but its still not working 100% of the time was wondering if there are any reasons for this. I have a 5v to 3.3v voltage devidor from mega tx to esp rx and a 3.3v to 5v step up diode  circuit from the esp to the mega so im not sure if its the hardware

profisir, 2018-02-18:
First check log on EspLink website
Maybe You have error like this:
"avrdude: stk500v2_getsync(): timeout "

frizzle101101, 2018-02-18:
No its readpacket got1 0, expected 5...its like it gets in sync then starts
loosing data

profisir, 2018-02-19:
Try change Diode to small resistor like 100-470 Ohm