jeelabs/esp-link #314: Suppressing debug info

eporocrail, 2017-07-17:
I am looking at the MQTT possibilities. 

The question I have is how do I suppress the debug information:
ELC: got 38 @330: 3 273 2 14 0 2F 68 65 6C 6C 6F 2F 77 6F 72 6C 64 2F 61 72 64 75 69 6E 6F 0 0 2 0 31 37 A9 FD
RESP_CB: 627 2

uzi18, 2017-07-17:
Try ELClient esp(&Serial); 
instead of ELClient esp(&Serial, &Serial);

swilson86, 2017-07-18:
I have been hounded by those debug messages too! The solution above works. It would help newbies if the examples had a brief narrative such as:

/* Initialize a connection to esp-link using the normal hardware serial port both for
 SLIP and for debug messages. Debug messages (on the uC Consul) look something like this:
 ELC: got 34 @577: 3 4BC 2 D 0 48 72 6F 6E 74 34 6F 6F 72 2F 4C 44 
 RESP_CB: 1212 2
*/
// ELClient esp(&Serial, &Serial); // Use this for debug messages
ELClient esp(&Serial); // Use this to suppress debug messages

eporocrail, 2017-07-18:
Thanks guys
This solution works. I was more thinking about how to set "_debugEn".

closed 2017-07-18 by eporocrail