jeelabs/esp-link #315: Cannot set baud below 9600

mrburns0, 2017-07-24:
Hi there,

I'm having difficulty setting the baud rate below 9600 with both the latest alpha build (3.2.27), and version 3.0.14.  I've replicated on a few modules that I have (all from the same source, so can't rule out my batch being bad).  Not sure if anyone else has seen/can replicate?

Device flashes fine, and is fully functional until restarted then gets stuck in a boot loop.  

To replicate:  Flash module, set baud to anything less than 9600, restart device.

I can do any necessary testing if needed.

Thanks,
MB

uzi18, 2017-07-24:
Post some log from terminal,  isn't it bootloader bug?

mrburns0, 2017-07-24:
See below the UART output from a chip set to 2400 baud.  Incidentally it is actually communicating at 2400baud, but resets when trying to bring up the wifi.

This was a 'new' device flashed with version 3.0.14

----------

����xTZM򕱖I��E���HHH�����j
                                                   111> ** esp-link v3.0.14-g963ffbb
   111> Flash config restore ok
   111> CONN led=0
   129> SER led=14
   213> Wifi init, mode=AP+STA
   346> Wifi uses DHCP, hostname=esp-link
   525> "ip": "0.0.0.0"
   629> "netmask": "0.0.0.0"
   754> "gateway": "0.0.0.0"
   879> "hostname": "esp-link"
  1012> sleep enable,type: 2
  1137> Httpd init, conn=0x3fff3820
  1291> No user file system found!
  1441> Serbridge pins: reset=12 isp=13 swap=0
  2058> Reset cause: 6=external
  2058> exccause=0 epc1=0x0 epc2=0x0 epc3=0x0 excvaddr=0x0 depc=0x0
  2066> Flash map 4MB:512/512, manuf 0xE0 chip 0x4014
  2296> ** esp-link v3.0.14-g963ffbb: ready, heap=21368
  2533> SNTP timesource set to us.pool.ntp.org with offset 0
  2791> initializing user application
  2954> Waiting for work to do...
*�򰄭�)> mode : sta(2c:3a:e8:0b:a9:83) + soft��Є

   113> 
   113> ** esp-link v3.0.14-g963ffbb
   113> Flash config restore ok
   113> CONN led=0
   131> SER led=14
   214> Wifi init, mode=AP+STA
   347> Wifi uses DHCP, hostname=esp-link
   526> "ip": "0.0.0.0"
   631> "netmask": "0.0.0.0"
   756> "gateway": "0.0.0.0"
   881> "hostname": "esp-link"
  1014> sleep enable,type: 2
  1139> Httpd init, conn=0x3fff3820
  1293> No user file system found!
  1443> Serbridge pins: reset=12 isp=13 swap=0
  2060> Reset cause: 3=soft wdt
  2060> exccause=4 epc1=0x40102701 epc2=0x0 epc3=0x0 excvaddr=0x0 depc=0x0
  2097> Flash map 4MB:512/512, manuf 0xE0 chip 0x4014
  2326> ** esp-link v3.0.14-g963ffbb: ready, heap=21368
  2564> SNTP timesource set to us.pool.ntp.org with offset 0
  2822> initializing user application
  2985> Waiting for work to do...
  3131> mode : sta(2c:3a:e8:0b:a9:83)̈́��p��V�7)

<repeat>

tve, 2017-07-27:
I believe this has been answered elsewhere, but for the record, this most likely is due to the fact that TX at 2400 baud hogs the CPU for too long. UART TX would have to be interrupt driven for this to work. The TX may be debug messages, so you need to make sure that no debug output is printed to UART0 and it may work.

mrburns0, 2017-07-27:
That's exactly it!  Disabling the debug output on boot and resetting the chip to 2400 baud now lets it boot successfully.  

Thank you for your help.

closed 2017-07-27 by mrburns0