jeelabs/esp-link #318: Question ESP32 Port

kwoodbury1, 2017-07-26:
We need the Bluetooth connectivity in the ESP32.
Can the the esp-link project be ported to the ESP32?
How much work would it be?

Thanks!
Kerry

tve, 2017-07-27:
I have no experience with the ESP32, so I cannot comment on this, sorry.

Sergio-tix, 2020-02-11:
ESP32 more productive and has 3xUART hardware up to 5Mbps (from datasheet). It makes sense to port. In combination with Tibbo software, you can get a converter for several protocols at once, RS485/232/CAN applying cheap converters from Ali, as MAX3232.

uzi18, 2020-02-11:
yes, we know but someone need free time to port it

eried, 2021-01-18:
ESP32 is almost the same price as the ESP8266, has anyone started/tried to port esp-link?

tve, 2021-01-19:
Yeah... 'been using micropython on the esp32 for a year now, really liking it... it would make sense to port the UI to micropython on the esp32 and create a version of esp-link that is easily scriptable... but that does need time and motivation...

closed 2024-10-29 by dpulvermacher