jeelabs/esp-link #319: esp-link on a RobotDyn ATmega2560+ESP8266

helitron, 2017-07-27:
I have six Wemos D1 mini, flashed with esp-link, already in use successfully for flashing and debugging six Arduino mega 2560 remotely. For flashing I am using the script megaflash.sh on Windows 10. No problems at all.

Recently I have got two of this boards:

http://robotdyn.com/catalog/arduino/mega_wifi_r3_atmega2560_esp8266_flash_8gb_usb_ttl_ch340g_micro_usb/

This is a Arduino mega2560 clone with an ESP8266 on board already. The ESP8266 is equipped with a 25Q32 flash chip that means 32 MBit = 4MByte. I don't know why they write 8MB flash size on the website mentioned above but that's not the problem.

When I am loading via Arduino IDE the ESP8266 example sketch "WifiWebServer" to the ESP8266 on this board the expected SSID "ESP_xxxxxx" is popping up in the Wifi Scanner for some seconds. After that the example sketch is working as expected (LED can be switched on/off via Webbrowser). That means the ESP8266 is working basically.

But now the problem I have got: when I am flashing esp-link to the ESP8266 on the board there is no SSID "ESP_xxxxxx" to see in the Wifi Scanner, so it is impossible to connect esp-link with the local network. For flashing the ESP8266 I am using the well known NODEMCU FIRMWARE PROGRAMMER, all my ESP8266 with esp-link on board mentioned at the beginning I have flashed with this tool already.

For flashing esp-link I am using the following settings:

Baudrate 460800
Flash size 4MByte
Flash speed 80MHz
SPI Mode DIO

boot_v1.7.bin 0x00000
user1.bin 0x01000
esp_init_data_default.bin 0x3FC000
blank.bin 0x3FE000

I have also tried to change all the settings to other flash sizes (1MByte, 8Mbyte, 16Mbyte) but without success. As I have mentioned at the beginning already a 25Q32 flash chip is on board, so it should be a 4Mbyte chip.

Thanks in advance for any helpful hint.

Erwin

tve, 2017-07-27:
Are you sure it's not QIO?
In general, to troubleshoot why the AP doesn't show I'd need the debug output on uart1 or uart0.

helitron, 2017-07-27:
Thanks for the hint, will try it with QIO also.
Btw. debug output, have never used it, is it necessary to activate the debug anyhow ? After restart of the ESP8266 there is always some output on the USB but I can't read anything useful, tried all possible baudrates from 9600 to 921600.

helitron, 2017-07-27:
In the meantime I've found a baudrate to read the output of the ESP8266 on the USB after restart (with flashed esp-link), it's 74880 baud. Strange baudrate.

The output after restart is only some characters:

ets Jan  8 2013,rst cause:2, boot mode:(1,7)

That's all.

helitron, 2017-07-27:
For double check I just flashed the ESP8266 example "WifiWebServer" from the Arduino IDE into the ESP8266 of the RobotDyn board. Immediately after restart the SSID ESP_FD8BBC is popping up on the Wifi Analyzer and both implemented commands in the example sketch (LEDon, LEDoff) are executable via webbrowser.

But with flashed esp-link still no success unfortunately. In my opinion there must be any fault in the settings of the NODEMCU FIRMWARE PROGRAMMER but as I mentioned already with the same settings I've flashed esp-link to six Wemos D1 mini successfully.

uzi18, 2017-07-28:
try esptool

helitron, 2017-07-28:
@uzi18 thanks again for the hint with esptool

Just tried it with esptool, no problem with flashing, a lot of output (see below) but unfortunately still no SSID in Wifi Analyzer.

D:\3.2.47>esptool.py --port COM3 write_flash 0x00000 boot_v1.7.bin 0x01000 user1.bin 0x3fc000 esp_init_data_default.bin 0x3fe000 blank.bin
esptool.py v2.1-beta1
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Flash params set to 0x0040
Compressed 4080 bytes to 2936...
Wrote 4080 bytes (2936 compressed) at 0x00000000 in 0.3 seconds (effective 124.1 kbit/s)...
Hash of data verified.
Compressed 344708 bytes to 253827...
Wrote 344708 bytes (253827 compressed) at 0x00001000 in 22.4 seconds (effective 123.2 kbit/s)...
Hash of data verified.
Compressed 128 bytes to 75...
Wrote 128 bytes (75 compressed) at 0x003fc000 in 0.0 seconds (effective 78.8 kbit/s)...
Hash of data verified.
Compressed 4096 bytes to 26...
Wrote 4096 bytes (26 compressed) at 0x003fe000 in 0.0 seconds (effective 4681.2 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting...

D:\3.2.47>

uzi18, 2017-07-28:
Do you have any dip switch on mega board?
Is is possible to pass through serial output from esp to usb converter?

Try to catch debug log from esp and put it here

helitron, 2017-07-28:
The ESP IS connected at the moment directly to the USB chip (yes via DIP switches), this is necessary to program the ESP. And the debug output  is only this single line (I mentioned it already above):

ets Jan 8 2013,rst cause:2, boot mode:(1,6)

That's all, no further output after restart.

uzi18, 2017-07-28:
did You tried first erase complete flash with: esptool.py erase_flash ?

helitron, 2017-07-28:
No, I don't, will try it, thanks for the hint.

helitron, 2017-07-28:
Just erased the flash completely and reflashed esp-link with esptool, same behaviour as before unfortunately (no SSID on WiFi Analyzer).

Can't believe it that esp-link is not running on this ESP ... but at the moment I don't have any idea what I could try furthermore.

tve, 2017-07-28:
Can you please try release 3.0.14: https://github.com/jeelabs/esp-link/releases/tag/V3.0.14
There is a problem with 3.2.47 that I haven't been able to solve because I cannot reproduce it.

tve, 2017-07-28:
If you have a few minutes to spare and could try https://s3.amazonaws.com/s3.voneicken.com/esp-link/esp-link-v3.2.49-ge57f29c.tgz that would help me know whether that build is also broken or not. Thanks!

helitron, 2017-07-28:
Of course, no problem, will try it.

Btw. all my other (six) ESP8266/Wemos D1 mini are running with 3.2.47 without any problem.

uzi18, 2017-07-28:
Is it related to new SDK?

helitron, 2017-07-28:
Tried 3.0.14, flashing ok but same problem.

Same with new 3.2.49, here the output of esptool:

D:\3.2.49>esptool.py --port COM3 write_flash -fm qio -ff 80m 0x00000 boot_v1.7.bin 0x01000 user1.bin 0x3fc000 esp_init_data_default.bin 0x3fe000 blank.bin
esptool.py v2.1-beta1
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Flash params set to 0x004f
Compressed 4080 bytes to 2936...
Wrote 4080 bytes (2936 compressed) at 0x00000000 in 0.3 seconds (effective 124.1 kbit/s)...
Hash of data verified.
Compressed 343828 bytes to 253198...
Wrote 343828 bytes (253198 compressed) at 0x00001000 in 22.3 seconds (effective 123.4 kbit/s)...
Hash of data verified.
Compressed 128 bytes to 75...
Wrote 128 bytes (75 compressed) at 0x003fc000 in 0.0 seconds (effective 85.3 kbit/s)...
Hash of data verified.
Compressed 4096 bytes to 26...
Wrote 4096 bytes (26 compressed) at 0x003fe000 in 0.0 seconds (effective 4681.2 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting...

D:\3.2.49>

helitron, 2017-07-28:
For comparision I just tried to flash a brandnew Wemos D1 mini with the 3.2.49, was successful (as usual), esp-link ran immediately and the SSID "ESP_14E695" popped up on the WiFi Analyzer.

One alteration was necessary in the programming commandline though, the Wemos D1 mini is not running with -fm qio, it must be -fm dio.

Here the esptool output during programming the Wemos D1 mini:

D:\3.2.49>esptool.py --port COM3 write_flash -fm dio -ff 80m 0x00000 boot_v1.7.bin 0x01000 user1.bin 0x3fc000 esp_init_data_default.bin 0x3fe000 blank.bin
esptool.py v2.1-beta1
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Flash params set to 0x024f
Compressed 4080 bytes to 2936...
Wrote 4080 bytes (2936 compressed) at 0x00000000 in 0.3 seconds (effective 123.6 kbit/s)...
Hash of data verified.
Compressed 343828 bytes to 253198...
Wrote 343828 bytes (253198 compressed) at 0x00001000 in 22.2 seconds (effective 123.7 kbit/s)...
Hash of data verified.
Compressed 128 bytes to 75...
Wrote 128 bytes (75 compressed) at 0x003fc000 in 0.0 seconds (effective 85.3 kbit/s)...
Hash of data verified.
Compressed 4096 bytes to 26...
Wrote 4096 bytes (26 compressed) at 0x003fe000 in 0.0 seconds (effective 4095.9 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting...

D:\3.2.49>

Unfortunately I don't have a schematic of this RobotDyn board, so I don't know the exact connections of the ESP8266 on the board. But, I mentioned it multiple times already, the "WifiWebServer" sketch of the ESP8266 examples on the Arduino IDE is running flawlessly.

tve, 2017-07-28:
I'm afraid something else must be the problem. I don't see why it would work on one module and not the other if it's not a HW problem... Maybe power supply related, maybe related to the way one of the pins is hooked-up. Something you could try is to pass `--after soft_reset` as option esptool, this way it doesn't do a hard reset. If that works you have a problem with gpio0/2/15 that prevent a proper reboot into the firmware.

tve, 2017-07-28:
(If it's any comfort, over the years I have wasted days with flashing and reboot issues with the esp8266. It's really horrid in this respect, way worse than any other uC I know...)

helitron, 2017-07-28:
One problem is solved but not all problems unfortunately.

In the meantime I've found a tiny switch on the board that have connected the GPIO0 constantly with LOW. After flipping the switch esp-link is starting at last. Don't know why the "WifiWebServer" example sketch was able to start with the GPIO0 at LOW state.

Ok, esp-link is starting now and the SSID is also popping up on the Wifi Analyzer so I was able to connect the ESP8266 with my Android phone via Wifi. But when I try to load the esp-link WebUI only the basic home page is popping up with the background and some characters. The browser is not able to load all the values for the homepage from the ESP. The browser shows only some red turning circles. Here the complete log for the descripted sequence:

Boot baudrate 74880:
==============

ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 2592, room 16
tail 0
chksum 0xf3
load 0x3ffe8000, len 764, room 8
tail 4
chksum 0x92
load 0x3ffe82fc, len 676, room 4
tail 0
chksum 0x22
csum 0x22

2nd boot version : 1.7(5d6f877)
SPI Speed : 80MHz
SPI Mode : DIO
SPI Flash Size & Map: 32Mbit(512KB+512KB)
jump to run user1 @ 1000

rf cal sector: 128
freq trace enable 0
rf[112] : 00
rf[113] : 00
rf[114] : 01

SDK ver: 2.1.0(116b762)

From here baudrate 115200:
==================

124>
   124> ** esp-link v3.0.14-g963ffbb
   124> Flash config restore *FAILED*
   125> CONN led=0
   125> SER led=14
   127> Wifi init, mode=AP
   129> Wifi uses DHCP, hostname=esp-link
   133> "ip": "0.0.0.0"
   135> "netmask": "0.0.0.0"
   138> "gateway": "0.0.0.0"
   140> "hostname": "<null>"
   143> sleep enable,type: 2
   145> Httpd init, conn=0x3fff3820
   149> No user file system found!
   152> Serbridge pins: reset=12 isp=13 swap=0
   165> Reset cause: 6=external
   165> exccause=0 epc1=0x0 epc2=0x0 epc3=0x0 excvaddr=0x0 depc=0x0
   165> Flash map 4MB:512/512, manuf 0xC8 chip 0x4016
   170> ** esp-link v3.0.14-g963ffbb: ready, heap=21368
   175> SNTP timesource set to us.pool.ntp.org with offset 0
   180> initializing user application
   183> Waiting for work to do...
   186> mode : softAP(5e:cf:7f:fd:8b:bc)
   190> add if1
   192> dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
   199> bcn 100
 15145> Wifi check: mode=AP status=255
 32640> add 1
 32640> aid 1
 32641> station: 38:aa:3c:79:e1:d8 join, AID = 1
 32641> Wifi AP: station 38:aa:3c:79:e1:d8 joined, AID = 1
 32955> err already associed!
 32956> station: 38:aa:3c:79:e1:d8 leave, AID = 1
 32956> rm 1
 32956> Wifi AP: station 38:aa:3c:79:e1:d8 left, AID = 1
 33270> add 1
 33270> aid 1
 33271> station: 38:aa:3c:79:e1:d8 join, AID = 1
 33271> Wifi AP: station 38:aa:3c:79:e1:d8 joined, AID = 1
 37103> err already associed!
 37103> station: 38:aa:3c:79:e1:d8 leave, AID = 1
 37104> rm 1
 37104> Wifi AP: station 38:aa:3c:79:e1:d8 left, AID = 1
 40623> add 1
 40623> aid 1
 40623> station: 38:aa:3c:79:e1:d8 join, AID = 1
 40624> Wifi AP: station 38:aa:3c:79:e1:d8 joined, AID = 1
178850> HTTP GET /pins: 200, 9786ms, h=7624
179100> HTTP GET /menu: 200, 9993ms, h=7696
179850> HTTP GET /system/info: 200, 10752ms, h=9680
179851> HTTP GET /wifi/info: 200, 10779ms, h=9920

uzi18, 2017-07-29:
now You can try erase flash and flash with latest alpha :)

helitron, 2017-07-29:
Will do it, thanks :)

helitron, 2017-07-29:
Erased flash and loaded 3.2.49 to the ESP. Connection is possible to my Android phone but extreme slow, it's not possible to load the esp-link homepage. Tried it with three different Android devices.

Here the log output:

130>
   130> ** esp-link v3.2.49-ge57f29c
   130> Flash config restore *FAILED*
   130> CONN led=0
   131> SER led=14
   133> Wifi init, mode=AP
   135> Wifi uses DHCP, hostname=esp-link
   139> "ip": "0.0.0.0"
   141> "netmask": "0.0.0.0"
   143> "gateway": "0.0.0.0"
   146> "hostname": "<null>"
   148> sleep enable,type: 2
   151> Httpd init, conn=0x3fff40c8
   154> No user file system found!
   157> Serbridge pins: reset=12 isp=13 swap=0
   170> Reset cause: 6=external
   170> exccause=0 epc1=0x0 epc2=0x0 epc3=0x0 excvaddr=0x0 depc=0x0
   171> Flash map 4MB:512/512, manuf 0xC8 chip 0x4016
   175> ** esp-link v3.2.49-ge57f29c: ready, heap=19080
   180> SNTP timesource set to us.pool.ntp.org with offset 0
   186> initializing user application
   189> Waiting for work to do...
   192> mode : softAP(5e:cf:7f:fd:8b:bc)
   196> add if1
   197> dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
   205> bcn 100
 12440> add 1
 12440> aid 1
 12440> station: 38:aa:3c:79:e1:d8 join, AID = 1
 12440> Wifi AP: station 38:aa:3c:79:e1:d8 joined, AID = 1
 15151> Wifi check: mode=AP status=255
 84506> HTTP GET /: 302, 170ms, h=16240
 84774> HTTP GET /home.html: 200, 436ms, h=15920
 94761> HTTP GET /ui.js: 200, 9917ms, h=11872
 95011> HTTP GET /style.css: 200, 10178ms, h=11968
 95012> HTTP GET /pure.css: 200, 10186ms, h=12264
118827> HTTP GET /favicon.ico: 200, 138ms, h=15048

helitron, 2017-07-29:
In the meantime I tried another two tablets, connection with 192.168.4.1 is possible but not to load the esp-link homepage with the browser (tried Android standard browser and Goggle Chrome). The browser tries to load the page for a long time but is not able to show anything.

Each try to load the homepage ends with this log on the USB:

749888> station: cc:79:cf:08:df:a2 join, AID = 1
749888> Wifi AP: station cc:79:cf:08:df:a2 joined, AID = 1
877356> HTTP GET /: 302, 10009ms, h=14256
879606> HTTP GET /home.html: 200, 9923ms, h=14784
979750> HTTP GET /: 302, 9826ms, h=13704
981749> HTTP GET /home.html: 200, 9956ms, h=13256
288405> HTTP GET /home.html: 200, 9818ms, h=12880
312978> HTTP GET /favicon.ico: 200, 9857ms, h=12152
312979> HTTP GET /home.html: 200, 9807ms, h=12360

Btw. on all Android devices the Wifi Analizer shows a very good RSSI of -35 of the ESP8266, so the Wifi connection should be excellent.

In the meantime I tried also some other ESP8266 network examples from the Arduino IDE on the same ESP8266 (RobotDyn board), each of the examples is working flawlessly, no problem at all.

tve, 2017-08-04:
The log shows some really slow page generation times, >9 seconds. I've seen this on occasion but I have no idea what causes it.

helitron, 2017-08-04:
Ok, thanks Thorsten. Played around some additional hours with the board but gave then up. It's a pity, was a very promising board with the combination of the mega and the ESP.

ristomatti, 2017-10-26:
@helitron I haven't tried esp-link yet myself but I own the same RobotDyn board. I was unable to flash anything on the ESP8266 it has onboard. I did not have that issue with another similar RobotDyn board UNO+WiFi. I ordered the "same" Mega board with Wems branding and have had no issues flashing some Arduino sketches into it. So I believe the ESP8266 on my RobotDyn Mega board is faulty and that might also be the case for you.

I'll try to find time to flash esp-link on the Wems branded board later this week as I got curious to try esp-link also. If I do, I'll follow up with the results. It's good to know D1 Mini works as an alternative as I have a few of them also. :)

helitron, 2017-10-26:
@ristomatti 
Flashing the ESP8266 on the Robodyn Mega board was never a problem for me (see thread from the beginning), my problem was always that eps-link flashed on the ESP8266 cannot connected with a phone or tablet (SSID "ESP_xxxxxx" is not popping up). But when I flash the Arduino example "WifiWebServer" to the ESP8266 it is working as expected (SSID pops up and it possible to connect a phone or tablet).

ristomatti, 2017-10-27:
@helitron To me that would only mean it can be flashed but not that it is 100% working correctly.

uzi18, 2017-10-27:
Did You tryied to erase complete flash?

helitron, 2017-10-27:
@ristomatti 
exactly

helitron, 2017-10-27:
@uzi18 
Yes I did

ristomatti, 2017-10-27:
@helitron I gave this a go now and had similar results. I tried flashing both v3.0.14 and v3.2.47 with esptool:
```
esptool.py write_flash 0x00000 boot_v1.7.bin 0x01000 user1.bin 0x3fc000 esp_init_data_default.bin 0x3fe000 blank.bin
```

With both versions the AP appears, I was able to connect to it and receive IP 192.168.4.2 with gateway 192.168.4.1. The gateway IP responded to pings but there was no web server listening on port 80.

Serial console tells me this:
```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 2592, room 16 
tail 0
chksum 0xef
load 0x00000000, len 0, room 8 
tail 0
chksum 0xef
load 0x00000000, len 0, room 0 
tail 0
chksum 0xef
csum 0xef
csum err
ets_main.c 
```

However I was not able to get this to work with a Wemos D1 Mini or Wemos D1 Mini Pro boards. Did I miss some step in the flashing process?

ristomatti, 2017-10-27:
I forgot to mention that I also tried flashing `HelloServer.ino` web server test sketch to the ESP onboard the Mega2560 board and it seemed to work as expected.

What I have been doing with these RobotDyn combo boards currently is flash a sketch that redirects JSON send to the onboard ESP through serial. I've got the boards in good use like this but esp-link seems like a much more convenient option (if I could get it to work).

In case someone is interested the undocumented quickly hacked together code I'm using for this can be found here [ristomatti/esp8266-rxtx-mqtt](https://github.com/ristomatti/esp8266-rxtx-mqtt).

MiloshCZ, 2017-11-01:
I have found correct setting for this board:
`python "../esptool/esptool.py" --port COM33 --baud 115200 write_flash --flash_freq 80m --flash_mode qio --flash_size 1MB 0x0000 boot_v1.7.bin 0x1000 user1.bin 0xFC000 esp_init_data_default.bin 0xFE000 blank.bin`
change COM33 to your COM port number

gheesung, 2017-12-01:
Recently found this interesting board. I have managed to set up the board using the recommended [flashing parameters](https://github.com/jeelabs/esp-link/blob/master/FLASHING.md)  for the ESP 8266.  I am using the 3.0.14 firmware. I have also created a simple sketch using APDS sensor on Mega2560 to send ambience and RGB light data to MQTT

I think the trick is in the dip switch setting at each stage. I have done a short write-up on this.
http://iotdiary.blogspot.sg/2017/11/esp-link-on-atmega2560esp8266-board.html

ristomatti, 2017-12-05:
@gheesung I'm afraid it might be you've just been lucky to receive a unit with an ESP that works properly. I tried once again, this time flashing with the parameters described by @MiloshCZ. I am able to flash the board but on boot I just see this on serial at 76800 baud:
```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 2408, room 16 
tail 8
chksum 0xef
load 0x00000000, len 0, room 0 
tail 0
chksum 0xef
load 0x00000000, len 0, room 8 
tail 0
chksum 0xef
csum 0xef
csum err
ets_main.c 
```
...and I'm using the unit that I've got working by flashing my own Arduino based code. The other RobotDyn Mega+WiFi unit I was never able to flash in any way. Even `esptool chip_id` never connects on that one.

gheesung, 2017-12-06:
@ristomatti so sorry that yours was not working properly. My board is not Robotdyn, it is the other brand that is selling on Aliexpress. Initially, I had tried all the above methods including @MiloshCZ method (btw these flashing parameters is for esp-01 which I have another setup using ESP-01 and Pro Mini), it did not work. Later I realised it was the dip switch 7 is still in ON position. After I switch to OFF, everything works.

You can also try flashing the firmware by directly connecting to the ESP pin header using an external FTDI adapter. I did that for Mongoose OS which refused to flash via the USB port. For this method, you have to set the DIP switches to all OFF (independent mode) before flashing.

drmatrix, 2017-12-14:
hei
im facing the the problem that the name of chip have change to esp8266ex . and nothing is working anymore

that its my plash process

C:\python\esptool\esptool>esptool.py --port COM8 --baud 115200 write_flash -fs 32m -ff 80m  0x00000 boot_v1.6.bin 0x1000 user1.bin  0x3FC000 esp_init_data_default.bin 0x3FE000 blank.bin
WARNING: Flash size arguments in megabits like '32m' are deprecated.
Please use the equivalent size '4MB'.
Megabit arguments may be removed in a future release.
esptool.py v2.2
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash params set to 0x004f
Compressed 3856 bytes to 2763...
Wrote 3856 bytes (2763 compressed) at 0x00000000 in 0.2 seconds (effective 123.9 kbit/s)...
Hash of data verified.
Compressed 344708 bytes to 253827...
Wrote 344708 bytes (253827 compressed) at 0x00001000 in 22.5 seconds (effective 122.8 kbit/s)...
Hash of data verified.
Compressed 128 bytes to 75...
Wrote 128 bytes (75 compressed) at 0x003fc000 in 0.0 seconds (effective 78.8 kbit/s)...
Hash of data verified.
Compressed 4096 bytes to 26...
Wrote 4096 bytes (26 compressed) at 0x003fe000 in 0.0 seconds (effective 4095.9 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting...

gheesung, 2017-12-14:
I flash with this parameters:

esptool --port com26 --baud 230400 write_flash -fm dio -fs 4MB -ff 80m 0x0 boot_v1.6.bin 0x1000 user1.bin 0x3fc000 esp_init_data_default.bin 0x3fe000 blank.bin

ristomatti, 2017-12-14:
I was finally able to succesfully flash esp-link to the ESP8266 on this board and also to a Wemos D1 Pro Mini few days ago. The parameters I used were:
```
esptool.py --baud 115200 write_flash --flash_freq 80m --flash_mode dio --flash_size 4MB 0x0000 boot_v1.6.bin 0x1000 user1.bin 0xFC000 esp_init_data_default.bin 0xFE000 blank.bin
```
The parameters don't seem to make any sense when reading again though as the addresses used for esp_init_data_defaullt.bin and blank.bin are the ones for an 8Mbit module based on the README.md included with ESP8266_NONOS_SDK. I can see from bash history that priot to that I've first run:
```
esptool.py write_flash 0xfc000 esp_init_data_default.bin
```
...and then
```
esptool.py write_flash 0x3fc000 esp_init_data_default.bin
```
The esp_init_data_default.bin being the one included with the ESP8266_NONOS_SDK v1.5.4 I had downloaded earlier on.

I should probably erase the flash again and retry to see what the actual parameters were. This is what you get when you try to make something work with more or less trial-and-error approach while not making any notes...

ristomatti, 2017-12-14:
@drmatrix ESP8266EX is printed also on the chips on both my ATmega2560+ESP8266 and Wemos D1 Mini Pro boards. Based on quick google it should be equivalent to an ESP8266E12

DJManas, 2018-04-07:
Sorry guys to „overtake“ this thread, but what do you use for Robotdyn Arduino Uno, which uses ESP-07 as pin setup?

uzi18, 2018-04-09:
@ristomatti Nice to hear it, some users had problem with flashing and maybe it is good hint for them.

SIACOMPUTACION, 2018-04-27:
Buenas 
He utilizado la placa ATmega2560+ESP8266 y tenia el mismo problema, lo resolvi de la siguiente manera
cargar boceto webserver en el esp8266  identificando la placa como NODEMCU1.0(ESP12-E MODULE)
Configurar en Flash size en "4M(1M SPIFFS)"  la opcion   ERASE FLASH "SKETCH + WIFI SETTINGS"
y cargar el webserver , los pines 5,6,7 en ON y los demas en OFF  
vereis que aparece el AP.
En caso de quitar la corriente a la placa es posible que el ap desaparezca, pero aparecera si poneis los pines 1,2,3,4 en ON y el resto en OFF  y el tipo de placa como ARTMEGA. y podeis cargar el boceto en artmega.
Yo lo he configurado con SPIFF para guardar la pagina web de control, en el ESp8266  como AP y poder acceder a traves de la ip 192.164.4.1  y cargar mi pagina web desde el webserver integrado en el ESP8266 , y en la placa ARTMEGA el biceto necesrio para controlar motores, luces led, y luces RGB

Espero  les sirva
Un Saludo a Todo

SIACOMPUTACION, 2018-04-27:
Disculpad la IP es la 192.168.4.1  que es la que viene por default

Parashutik, 2018-06-05:
Hello! I use your development for firming Arduino through Wifi. I have a separate block ESP8266-01. RX-TX pins are connected. And the contact resets Arduino to GPIO0. On Arduino Pro, Mini, Nano, Uno, everything works fine. Controllers flash without problems. But this does not happen on MEGA2560. Raises an error about timeout. If the reset button is pressed when the sketch download starts, the firmware is executed without any problems. It turns out that there is no reset of the controller before flashing through GIPO0. ESP8266 regularly unambiguously. There are no problems with other controllers. And I have several. Do not tell me what to do. ESP stitched with seven firmware what to eat. The latter costs 3.2.47 alpha

Parashutik, 2018-06-05:
So Arduino Mega2560 can not be used, using ESP8266-01, programming through Arduino IDE, like nano and others? Only through the script?

uzi18, 2018-06-06:
Exactly I use megaflash script for this. 
Do you have any other gpio available?

Parashutik, 2018-06-06:
There are only two of them on ESP8266-01. Second GPIO2. I've already tried it.

uzi18, 2018-06-06:
So try now megaflash script

Parashutik, 2018-06-06:
I understood. We will torment. :-)

Parashutik, 2018-06-12:
Если я пишу такую команду "curl --data '' http://192.168.5.15/console/reset", то MEGA перезагружается. Если я использую megaflash  скрипт, то выдает "./megaflash.sh 192.168.5.15 sketch_jun12a.ino.hex "

Parashutik, 2018-06-12:
If I write this "curl -data" command "http://192.168.5.15/console/reset", then MEGA restarts.
 If I am using a megaflash script, it issues a 
"Kolobaxa @ DOMASHNIY ~$ ./megaflash.sh 192.168.5.15 C: /Temp/arduino_build_853133/sketch_jun122.ino.hex
Error resetting AVR
"

Parashutik, 2018-06-12:
When sending in the browser command: http://192.168.5.15/pgmmega/synck writes: "Not Found."

Parashutik, 2018-06-12:
The same and the terminal

uzi18, 2018-06-12:
Try this in the browser:
http://192.168.5.15/pgmmega/sync

Are you using cygwin? 

./megaflash.sh 192.168.5.15 C:/Temp/arduino_build_853133/sketch_jun122.ino.hex

You have space after C: please remove it

Parashutik, 2018-06-12:
Yes, sorry, I forgot to say. I use Windows and Qigwin. Now remove the space and try.

Parashutik, 2018-06-12:
Removed the space, the error is the same. Error resetting AVR.   http://192.168.5.15/pgmmega/sync writes: "Not Found."

Parashutik, 2018-06-12:
bat file with the following content:
curl --data '' http://192.168.5.15/console/reset
SET ADDRESS = 192.168.5.15
SET BAUD = 115200
SET HEXFILE = C: \ Temp \ arduino_build_657958 \ pr1.ino.hex
set prog = D: \ Arduino \ hardware \ tools \ avr / bin / avrdude
set conf = D: \ Arduino \ hardware \ tools \ avr / etc / avrdude.conf
% prog% -C% conf% -v -p atmega2560 -b% BAUD% -c wiring -P net:% ADDRESS%: 23 -D -Uflash: w:% HEXFILE%: i

  It's all very nice.

[6 more replies]