jeelabs/esp-link #322: how to change default IP address and port number?

ulinktech, 2017-08-03:
hi 
i wanna change the default ip address and port. Now ip address is 192.168.4.1:23.
but can not change at http server. 
How to change it? 
My means that i want change IP address and port of esp8266 module.
pls inform to me.

regards

uzi18, 2017-08-03:
Why can't You change it by http, it doesn't work?

ulinktech, 2017-08-03:
yes. we did changed it by http of static ip address then saved  but can't change it.
also cannot found change function of port number.

uzi18, 2017-08-04:
Just reflash default options.

tve, 2017-08-04:
Changing the IP address of the AP is not supported. Most likely there is an SDK call you can insert somewhere to change it, but esp-link doesn't currently do that.

kriegste, 2017-08-04:
Why would anyone want to change the AP IP anyway?

ulinktech, 2017-08-04:
Our special apps are requires special IP and PORT. So, we have to change ip&port on esp8266 module.
But official  firmware was support ip&port change function. but official firmware was not support transparent function at server mode. how to ?

kriegste, 2017-08-04:
There is an example in the SDK.

```
wifi_set_opmode(STATIONAP_MODE);

struct ip_info info;

wifi_softap_dhcps_stop();
 
IP4_ADDR(&info.ip, 10, 10, 10, 1); 
IP4_ADDR(&info.gw, 10, 10, 10, 1); 
IP4_ADDR(&info.netmask,  255, 255, 255, 0); 
wifi_set_ip_info(SOFTAP_IF, &info);

wifi_softap_dhcps_start();
```

You will have to implement that by yourself.

The problem with the AP IP is that if you lose it, there is no fallback. You can just "guess" what it was. The better solution would be access by hostname (even in AP mode) but I have no clue how to do that...

ulinktech, 2017-08-04:
thanks for your reply. 
Let me take your suggestion . 
if any idea, pls inform to me.