jeelabs/esp-link #323: Problem with network upload on AVR 32u4

saspol, 2017-08-06:
Hello,
I setup chinese Nano 3.0 board with esp-link controller and everything works like charm. Then i tried to replace Nano with chinese 'micro' boards on 32u4 chip and none of them is accept firmware upload. I tried change port speed etc but no any success. If i manual telnet to upload port, i see there console output from current 32u4 firmware. Any suggestion please.

Linking .pioenvs/micro/firmware.elf
Checking program size
text       data     bss     dec     hex filename
13658       692     913   15263    3b9f .pioenvs/micro/firmware.elf
Building .pioenvs/micro/firmware.hex
Looking for upload port...
Use manually specified: net:192.168.98.33:23
Forcing reset using 1200bps open/close on port net:192.168.98.33:23
Waiting for the new upload port...
Error: Couldn't find a board on the selected port. Check that you have the correct port selected. If it is correct, try pressing the boar
d's reset button after initiating the upload.
*** [upload] Explicit exit, status 1

tve, 2017-08-06:
Please use the avrflash or megaflash scripts to troubleshoot the flashing of an attached AVR. Once those work you can decide whether you want to use the IDE's flashing instead.

saspol, 2017-08-06:
Whith avrflash i see something like this:

`  Waiting for sync...
* About to connect() to 192.168.98.33 port 80 (#0)
*   Trying 192.168.98.33...
* connected
* Connected to 192.168.98.33 (192.168.98.33) port 80 (#0)
> GET /pgm/sync HTTP/1.1
> User-Agent: curl/7.24.0 (x86_64-apple-darwin12.0) libcurl/7.24.0 OpenSSL/0.9.8} zlib/1.2.5
> Host: 192.168.98.33
> Accept: */*
> 
* HTTP 1.0, assume close after body
< HTTP/1.0 200 OK
< Server: esp-link
< Connection: close
< Cache-Control: no-cache, no-store, must-revalidate
< Pragma: no-cache
< Expires: 0
< 
{ [data not shown]
* Closing connection #0
Error checking sync: sync abandoned after 8 attempts`

And this with megaflash:

`Resetting AVR with http://192.168.98.33/pgmmega/sync
* About to connect() to 192.168.98.33 port 80 (#0)
*   Trying 192.168.98.33...
* connected
* Connected to 192.168.98.33 (192.168.98.33) port 80 (#0)
> POST /pgmmega/sync HTTP/1.1
> User-Agent: curl/7.24.0 (x86_64-apple-darwin12.0) libcurl/7.24.0 OpenSSL/0.9.8} zlib/1.2.5
> Host: 192.168.98.33
> Accept: */*
> 
* HTTP 1.0, assume close after body
< HTTP/1.0 404 Not Found
< Connection: close
< Content-Type: text/plain
< Content-Length: 12
< 
{ [data not shown]
* Closing connection #0
Error resetting AVR`

Have esp-link v2.2.3 on host

uzi18, 2017-08-07:
To use megaflash You need to update firmware first to latest v3.2.x

saspol, 2017-08-16:
I update latest 3.2.47, but everything is same.
megaflash show me infinite pages of this:

```
*   Trying 192.168.98.35...
* TCP_NODELAY set
* Connected to 192.168.98.35 (192.168.98.35) port 80 (#0)
> GET /pgmmega/sync HTTP/1.1
> Host: 192.168.98.35
> User-Agent: curl/7.54.1
> Accept: */*
> 
* HTTP 1.0, assume close after body
< HTTP/1.0 200 OK
< Server: esp-link
< Connection: close
< Cache-Control: no-cache, no-store, must-revalidate
< Pragma: no-cache
< Expires: 0
< 
{ [9 bytes data]
* Closing connection 0
  Waiting for sync...
```

I found Leonardo boards with 32u4 have two serial interfaces, like described here:
```
The Arduino Leonardo board uses Serial1 to communicate via TTL (5V) serial on pins 0 (RX) and 1 (TX). Serial is reserved for USB CDC communication. For more information, refer to the Leonardo getting started page and hardware page.
```

So, i can see console messages in esp-link web interface when print from MCU to Serial1, but not more than this. Any kind of firmware update does not working.

uzi18, 2017-08-16:
I assume Leonardo only support usb cdc programming.

saspol, 2017-08-18:
Its definately have programming via UART, here is example:
http://www.instructables.com/id/How-to-upload-sketches-to-Pro-MicroLeonardo-via-se/
I tried flash optiboot, and connect MCU RX via voltage divider, but still no success.

uzi18, 2017-08-18:
> Pro-MicroLeonardo-via-se/

Ok now I see,  you change bootloader.

Did you checked with usb/serial converter if you are able to upload basic
sketches e.g. Blink?

Have also some problems with optiboot on esp-link still investigate it.

saspol, 2017-08-18:
About flashing serial have also this information, but i dont know how i can use it.

> 
> Leonardo, Pro Micro, Micro, Robot, Esplora, have an ATmega32U4 processor. It has pins for the USB protocol and has low-level logic designed specifically to interface with USB. What this means is there is no USB-to-serial adapter or chip at all, and the ATmega32U4 is connected directly to USB of your computer.
> 
> The information is passed between the computer and the Arduino using a protocol that mimics or stands in for TTL serial, but there are no exposed pins for RX and TX involved. The RX and TX pins on the Leonardo and other ATmega32U4-based Arduinos are not involved at all in the bootloader or upload process. The USB connection is a kind of firmware-implemented UART0 that is used in your sketches with statements like Serial.print. The RX and TX pins are connected to a secondary UART, which is UART1, and are used in your sketch with statements like Serial1.print.
> 
> Since the USB CDC-serial interface does not have physical pins such as RX and TX and the other related TTL-232 pins, we don't have access to a DTR pin to reset the Arduino and make it run the bootloader. The USB pins are simply D+ and D-. What do we do about that? The Leonardo designers thought long and hard about this and came up with an interesting work-around. Although the CDC-serial interface in the ATmega32U4 processor does not really use baud rates like serial lines do, it is aware of when the computer is requesting to set a particular baud rate. They programmed the Arduino IDE to do a baud rate change to 1200 for just a second and then change it to 57600. This is called a 1200bps_touch and you can see the setting if you study the boards.txt file that is part of the Arduino IDE software package. The Leonardo detects this and resets the processor using a watchdog method. Then the bootloader runs. The bootloader is called Caterina. It is different from the ATmega328P-based Arduinos in that it does not accept the sketch using the RX and TX lines. Caterina uses the USB D+ and D- lines.
> 
> So, the Leonardo and other ATmega32U4-based Arduinos do not reset when you open serial monitor or start some other serial process with them. They only reset when the IDE does the 1200bps_touch just before uploading a sketch. That is why you don't need to disable automatic reset with a capacitor when you use it as a programmer.

When i upload bootloader use command like this:
```avrdude -P /dev/tty.usbmodemfa141 -b57600 -c avrisp -p m32u4 -v -e -U flash:w:optiboot_atmega32u4.hex -U lock:w:0x0F:m```
Bootloader blinking TX constantly tree times as expected.
Then i try upload same way Blink sketch for test Serial and Serial1 console in esp-link but looks like after this MCU not boot anymore. I cannot upload sketch from IDE because it overwriting optiboot.