jeelabs/esp-link #324: Major reliability issue

Serveurperso, 2017-08-12:
I tested many version of this firmware to make a simple transparent serial link to program my PIC32.
The auto-reset is not used because I handle myself with separate hardware.

But the reliability of the link is very bad, it slow down after few STK500 programming and ping timeout.
I tested the esp8266 and esp8266EX with same issue. I use breakout board with all necessary pullups/down onboard.
I can't use this project as a backup link to flash a firmware:(

uzi18, 2017-08-12:
Give us some hint,  from debug log after first upload and after 10th?

uzi18, 2017-08-12:
Stk500 v1 or v2?

Serveurperso, 2017-08-12:
Uploading with STK500v2 protocol 

(I need to remote-program my robot on this live page https://www.serveurperso.com/?page=robot all work but I need a real transparent and reliable serial bridge.)

My NodeJS process call :
/usr/bin/socat pty,link=/dev/ttyACM99,raw tcp:192.168.0.4:23
and 

"/usr/bin/uecide" +
                                 " --headless" +
                                 " --board=chipkit-max32" +
                                 " --core=chipkit" +
                                 " --compiler=pic32-tools" +
                                 " --port=/dev/ttyACM99" +
                                 " --programmer=pic32prog" +
                                 " --compile" +
                                 " --upload /var/www/robotadmin/Robot32/Robot32.ino"

I also tested the auto reset port 2323 with same slow-down-over-time and ping timeout problem of the esp-link firmware.

uzi18, 2017-08-12:
I'm more interested in debuglog from esp-link

Serveurperso, 2017-08-12:
yes I think but sorry I just flashed beckdac/ESP8266-transparent-bridge to check if more reliable

Serveurperso, 2017-10-30:
I re-flashed ESP-Link to re-test it. I'm ready to debug if I get problems.

Serveurperso, 2017-10-30:
I remplaced DC-DC power supply, I use new pic32prog from Matt (Majenko Technologies, removed the dynamic bootloader automatic baudrate detection) and now it look very reliable...

uzi18, 2017-10-30:
Good news =)
What dc/dc converter did you used?

Serveurperso, 2017-10-30:
For my 3.3V line on my project I used one anyvolt micro but it's bad -> lot of problem (and the famous ESP crash)

Two in parallel and tuned exactly to same voltage give reliable work (but temporary workaround)

Now I use the good one: "Isolated DC/DC Converters 9-36Vin 3.3Vout 2A 9W SIP Iso DC/DC"
https://eu.mouser.com/Power/DC-DC-Converters/Isolated-DC-DC-Converters/_/N-brwkv?P=1z0wuofZ1yx5k7vZ1yxt7bdZ1yxt7f5

Serveurperso, 2017-10-30:
look the ESP need a lot of power to get reliable else it give very strange and inconsistent working !
I think it make high current burst very short time. A good capacitor on power line is a good protection also.

Serveurperso, 2017-10-30:
If the ESP-Link ping all this day and I can work with my robot all this 3 days the issue is closed, else I upload logs...