jeelabs/esp-link #327: Problem getting this to work, Sync failed!

MartinTerp, 2017-08-19:
Hi guys.

Im still pretty new to arduino and im trying to connect my esp8266 running esp-link v2.2.3.
I have connected it using this schematic: https://i.stack.imgur.com/KYbW1.png

I have no problem connecting to the esp_link interface.

Im trying to use the REST example on the EL-Client lib.

- Arduino UNO
- ESPLINK baud: 115200
- SLIP Enabled

The message on the Serial Monitor:
`EL-Client starting!
ÀÀClient sync failed`

What am i missing?
Thanks in advance.

uzi18, 2017-08-19:
> KYbW1.png

try test-reset example, first
try also esp with latest firmware 3.x

MartinTerp, 2017-08-19:
@uzi18 Thanks for your response.

I have updated to 3.0.14, and tried running the test-reset example as you said, and im, getting the same error "ÀÀEl-Client sync failed"

Also, in the uc console, i get this when booting up:
`["#��a�q��������M�]�m�|��*�0�������������ߑϑ�����ߐϐ���������o���q�`����ϓߓ��� �0�g�p������������ U`

uzi18, 2017-08-19:
OK now try sketch like this:

void setup() {
Serial.begin(115200);
Serial.println("test");
Serial.println("test");
Serial.println("test");
Serial.println("test");
Serial.println("test");
}

upload it to uno

open uc console - choose 115200, 8n1
now please push reset on uno

did You received test lines in uc console?

if not You have wrong UNO selected or Your uno have got wrong crystal
selected.

MartinTerp, 2017-08-19:
Apprently i have switched the to rx/tx wires, worked after switching them around

closed 2017-08-19 by MartinTerp

uzi18, 2017-08-19:
So You didn't connected it like on image you attached.