jeelabs/esp-link #334: How to get bin file for flash

ffanez, 2017-09-16:
Hello everyone i am very lay about this and beginner.
I'm sorry to ask you, I use ESP8266-12 board will use firmware, i have downloaded this but i did not find the bin.file to be used flash to the board.
Can you guide me to get the bin.file of the downloaded results for my use.
Whether it can also be used for repeater menu with the same UI.
Thank you very much.

tve, 2017-09-16:
I have no idea what you downladed. Please start here: https://github.com/jeelabs/esp-link/releases/tag/V3.0.14 and download the tgz after the bold **Download:**

ffanez, 2017-09-16:
Thanks for the answer, I immediately follow your advice and I downloaded continued extract I can not find any file.bin ready for esp download tool?
I'm sorry I do not understand this, or do bin files have to be created by myself?

AdjunctExorcist, 2017-09-16:
Check under the releases section. You'll find prebuilt binaries there.

ffanez, 2017-09-17:
Thanks to his guides I've managed to flashing esp-link Release 3.0.14,
but can not AP + STA mode simultaneously, shortly after config AP lost no longer detected before the flash reset so on.

roesl, 2017-09-27:
is there a download for the bin files for Version 4.2.xx ?

tve, 2017-09-27:
All downloads are here: https://github.com/jeelabs/esp-link/releases
There is not version 4.2