jeelabs/esp-link #336: ESP-01S showing 512K of flash instead of 1M

rlerdorf, 2017-09-18:
I have a couple of ESP-01S 1M cards. Before flashing the serial console shows:

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 1856, room 16
tail 0
chksum 0x63                                                                                                                          
load 0x3ffe8000, len 776, room 8
tail 0
chksum 0x02
load 0x3ffe8310, len 552, room 8
tail 0
chksum 0x79
csum 0x79

2nd boot version : 1.5
  SPI Speed      : 40MHz
  SPI Mode       : DIO
  SPI Flash Size & Map: 8Mbit(512KB+512KB)
jump to run user1 @ 1000

rf cal sector: 249
rf[112] : 00
rf[113] : 00
rf[114] : 01

SDK ver: 1.5.4.1(39cb9a32) compiled @ Jul  1 2016 20:04:35
phy ver: 972, pp ver: 10.1
```

They flash ok with the latest 3.2.47:
```
$ esptool -cp /dev/ttyUSB0 -cb 460800 -cd none -bz 1M -ca 0x0000 -cf boot_v1.7.bin -ca 0x1000 -cf user1.bin -ca 0xFC000 -cf esp_init_data_default.bin -ca 0xFE000 -cf blank.bin

Uploading 4080 bytes from boot_v1.7.bin to flash at 0x00000000
....
Uploading 344708 bytes from user1.bin to flash at 0x00001000
.................................................................................................................................................................................................................................................................................................................................................
Uploading 128 bytes from esp_init_data_default.bin to flash at 0x000FC000
.
Uploading 4096 bytes from blank.bin to flash at 0x000FE000
....
```
But on boot esp-link shows:
```
301> Flash map 512KB:256/256, manuf 0xE0 chip 0x4014
306> ** esp-link v3.2.47-g9c6530d: ready, heap=18960
```

kriegste, 2017-09-19:
Have you made changes to your Makefile?

FLASH_SIZE ?= 1MB

rlerdorf, 2017-09-19:
No, I was actually just using the binary distribution from http://s3.voneicken.com/esp-link/esp-link-v3.2.47-g9c6530d.tgz
But I will try building my own and forcing the flash size there.