jeelabs/esp-link #339: Build from source on Linux fails

DouglasPearless, 2017-09-23:
A number of issues with the Makefile

(1) htmlcompressor URL has moved to https://storage.googleapis.com/google-code-archive-downloads/v2/code.google.com/htmlcompressor/htmlcompressor-1.5.3.jar

(2) when that is corrected in the makefile, then:
```
AR build/httpd_app.a
make: *** No rule to make target '/ld/eagle.app.v6.new.1024.app1.ld', needed by 'build/eagle.esphttpd1.v6.ld'.  Stop.

```

DouglasPearless, 2017-09-24:
An update, instead I have followed the docker instruction for Linux or Windows, but on OS X and I am able to build a fresh image from source :-)  

Cheers
Douglas

DouglasPearless, 2017-09-25:
Hmm, after the docker build there is no notes or indication how to load the resultant images :-(

tve, 2017-09-25:
The docker build is no different from a different build. The build tools run int he container, but the input & output are in your working directory. You end up with the same firmware subdir with the binaries and you load them the same way. There are several pages linked off the README.md that describe the options.

DouglasPearless, 2017-09-26:
OK, I built a new image using docker:

```
28 -rw-r--r-- 1 root root 27520 Sep 26 08:55 build/espfs.img
AR build/httpd_app.a
LD build/httpd.user1.out
Dump : /xtensa-lx106-elf/bin/xtensa-lx106-elf-objdump -x build/httpd.user1.out
Disass: /xtensa-lx106-elf/bin/xtensa-lx106-elf-objdump -d -l -x build/httpd.user1.out
FW firmware
build/httpd.user1.out  :
section            size         addr
.data              3210   1073643520
.rodata           19176   1073646736
.bss              39584   1073665912
.irom0.text      290212   1075843088
.text             31156   1074790400
Total            774855
    user1.bin uses 343828 bytes of 503808 available
LD build/httpd.user2.out
```
but I am unclear on how to load this image at the required location.

Previously I managed to load the pre-built images using [https://github.com/jeelabs/esp-link/blob/master/FLASHING.md](https://github.com/jeelabs/esp-link/blob/master/FLASHING.md)  :
```
32Mbit / 4Mbyte module

On Linux using esptool.py this turns into the following for a 32mbit=4MByte flash chip, such as an esp-12 module typically has (substitute the appropriate release number and bootloader version number):

curl -L https://github.com/jeelabs/esp-link/releases/download/v2.2.3/esp-link-v2.2.3.tgz | \
    tar xzf -
cd esp-link-v2.2.3
esptool.py --port /dev/ttyUSB0 --baud 230400 write_flash -fs 32m -ff 80m \
    0x00000 boot_v1.5.bin 0x1000 user1.bin \
    0x3FC000 esp_init_data_default.bin 0x3FE000 blank.bin
```
I must have missed something in the various instructions, how do  I determine how to load the new image to what location using esptool.py ?

Help :-)

DouglasPearless, 2017-09-27:
Managed to get it all sorted.

closed 2017-09-27 by DouglasPearless

milkpirate, 2022-01-04:
> Managed to get it all sorted.

Hey @DouglasPearless could you please share your solution?