jeelabs/esp-link #340: Question: How to edit the HTML page and rebuild entire project from source ?

Sumgit91, 2017-09-23:
Hi, thank you for your contribution, it's just a wonderful project, it works perfectly fine for my application. 
I want to change (edit) the HTML page, after doing so, we need to build the project right? Please, can anybody give a detailed description of the procedure and help me with the rebuilding of the project and get bin files.

DouglasPearless, 2017-09-24:
I too would like that as I cannot get the linux method to compile, and the docker method does not explain where to load the resulting files (in memory).

uzi18, 2017-09-24:
I'm able to compile under linux and interested in add some own html files.

I will try first add something more to rom, and give you feedback.

DouglasPearless, 2017-09-24:
When you (hopefully) have it working, could you please list clearly all the steps you followed?

Thanks in advance.

uzi18, 2017-09-24:
Ok , adding files is easy, just add file something.html or something.js to html dir and run "make"
But I understand You want to add them also into left side menu?

uzi18, 2017-09-24:
If You want to add something to menu, add it in esp-link/cgi.c file inside cgiMenu function

Sumgit91, 2017-09-27:
Thank you, @uzi18 will try that for sure.

Sumgit91, 2017-09-27:
@DouglasPearless. Hi, I'll explain the steps I followed for flashing ESP12.
Step1: download Node MCU flasher from here [](https://github.com/nodemcu/nodemcu-flasher) and run win32 or 64, whichever suits you.
Step2: Download Binaries (Bin files) of required version of ESP-Link you want.
Step3: Run Node MCU flasher, in operation menu select the COM port which your ESP is connected.
Step4: In Config menu give the paths of binary files in order and specify the addresses where you want to write them. In my case For ESP 12 it the following.
boot_v1.6.bin @ 0x00000
blank.bin @ 0x3FE000
default.bin @ 0x3FC000
user1.bin @ 0x01000
*make sure to mark the box beside the file window after adding.
Step4: In Advanced window for ESP 12 config the settings are as follows, flash size varies for different ESP's,
Baudrate: 460800
Flash Size: 4MByte
Flash Speed: 40MHz
SPI Mode: DIO

Step5: Reset the ESP module, connect GPIO 0 and 15 to ground, and GPIO 2 to 3.3 V at time of flashing, reset module before pressing the flash module.

Step6: connect GPIO 6 to Ground for normal boot operation.

closed 2017-12-09 by Sumgit91