jeelabs/esp-link #341: Can't configure WiFi from ESP01

RabeStudent, 2017-10-02:
First of all, thank u very much for this amazing project. I am using it for doing my own project using 2 ESP01.
The goal I have to reach is to give WiFi capabilities to 2 arduino boards (1 nano and 1 UNO). For connecting one with the other and transmit data.

I have flashed the firmwares to the 2 ESP01 I have and as I have read, the next step is to configure the WiFi of each one but when I connect my PC or my phone to anyone of the ESP's and try to load the page using IP (192.168.4.1) it does not work.

I don't know what is not working well or if I am skipping one step or something.
Can someone help me?

Thank u a lot!

uzi18, 2017-10-02:
Did you flash all files,  and choose correct flash version?
How big flash did esptool recognized?

tve, 2017-10-02:
The most reliable way to see what's going on is to look at the UART outputs after a HW reset using the normal TX and the gpio2 pins, both at 115200 baud.