jeelabs/esp-link #342: Reboot ESP-01 chip if Arduino detects WiFi failure

swilson86, 2017-10-04:
I have perhaps 18 ESP-01 chips with Arduinos running esp-client and esp-link. About once per week or so the esp-link/esp-client software will disconnect from WiFi (and it still sometimes responds to pings). I know the Arduino is still doing its' tasks due to LED responses to events, etc. If the Arduino could reset the ESP-01 chip when it detects it has gone crazy would solve a lot of problems. This is true in version 2.x and also 3.x of esp-client.

Anybody have experience in this area? Rebooting a crazy Arduino I can do, but the other way around? An Arduino sketch that checks to see if the esp-link has gone crazy would be ideal. I am thinking of the Reset pin on the ESP-01 chip and seeing if Arduino can force a reset. Ideas anyone?

Serveurperso, 2017-10-05:
For me ESP-Link never worked reliably on any hardware platform. There is slow-down like memory leak or any other not optimized trick

uzi18, 2017-10-05:
Turn on debug on uart in esp-link, and try to log it to file,  paste it
here we will try to investigate something

rjblake, 2017-10-19:
> For me ESP-Link never worked reliably on any hardware platform. There is slow-down like memory leak or any other not optimized trick

I have ESP-Link 3.0.14 running on my ESP connected to central heating system and it has worked without a problem for close to a year. The only time it has been restarted is when maintenance work was carried out on the boiler, so hardly ever restarted

Serveurperso, 2017-10-19:
For telemetry I think it work, but for STK500 flash over TCP there is a bad think.

uzi18, 2017-10-19:
But stk500 implementation is poor,  and could be improved.

Serveurperso, 2017-10-19:
STK500v2 is good and easy, and a transparent serial must be transparent - even a crappy protocol must not kill the radio firmware

swilson86, 2017-12-14:
I have not had a single need to reboot a crazy esp-link since I upgraded all my chips to 3.0.14 about 3 months ago. It's the Arduino that needs a reboot after running out of memory etc. Recommend this issue be closed.

closed 2017-12-14 by swilson86