jeelabs/esp-link #344: Serial - Telnet passthrough

DouglasPearless, 2017-10-24:
I have spent several weeks trying to diagnose and fix and issue and I have traced it back to ESP-Link.

Scenario: LaserWeb 4 <> AP <>Telnet ESP8266 running latest ESP-Link UART <> Smoothieboard.

The LaserWeb sends a command to Smoothie, Smoothie returns an"ok" or another response like "<Run ..." or "<Idle ..." back; this progresses nicely, then the message back to LaserWeb start getting corrupted and even have parts different responses get mixed up

Examplem from the ESP Console:
Was expecting:
```
X249.00 S0.6771
ok
```
Got
`X249.0 S0.6o771k`

HELP!

:-)

DouglasPearless, 2017-11-02:
So, I assume no-one else has this issue?
Not solvable?

uzi18, 2017-11-03:
Any screenshot?

DouglasPearless, 2017-11-16:
I am closing this issue as I have traced the root cause to the Smoothie part of the chain of events and not the ESP.

closed 2017-11-16 by DouglasPearless