jeelabs/esp-link #346: MDNS is only broadcasting over AP network

unlimitedbacon, 2017-10-28:
I can only get the MDNS broadcasts to work in AP or STA+AP mode. In STA+AP mode they are only broadcast over the AP side. It doesn't work at all in STA mode.

I looked through the code and couldn't find anything helpful. Is this an SDK issue?

kriegste, 2017-10-28:
There also needs to be a

wifi_set_broadcast_if(STATIONAP_MODE);

in your code. But I don't know if that solves your problem. Tell us.
MDNS is an ongoing nuisance. :(

unlimitedbacon, 2017-12-09:
That call is already present in socket.c
https://github.com/jeelabs/esp-link/blob/a59d4c3177763e6ff29f1fc24acb8387a82061ff/socket/socket.c#L299

I'm not sure where else it would go.

kriegste, 2017-12-09:
I am not sure if this is the case here, but

wifi_set_broadcast_if(STATIONAP_MODE); 

can only be called after STATIONAP_MODE is set. If you're in STATION_MODE or SOFTAP_MODE, the call will fail. At least this is what I found out in my experiments...

closed 2019-08-02 by tve