jeelabs/esp-link #348: Microcontroller Console - Clear Log - refresh/F5 - Do not clear the server side

Serveurperso, 2017-10-30:
Microcontroller Console - Clear Log - refresh/F5 - Do not work
The log is cleared only on client side, it need to be cleared on server side also.

uzi18, 2017-10-30:
You want to say it is cleared in browser not in esp-link cache?

Serveurperso, 2017-10-30:
I check this right now !

Serveurperso, 2017-10-30:
This is not the cache:( tested on different browser.
There is a loss of information also, and persistent unwanted information.
I do more tests

Serveurperso, 2017-10-30:
It's very buggy : Many information are sent to the client side, and volatile, disappear when I do F5.
Few information are freezed on the server side and never cleared when I push Clear Log.

Serveurperso, 2017-10-30:
tested on : Firefox & Chrome and fake browser like EDGE and IE.
tested also on Stock Android and Chrome for Android.

Serveurperso, 2017-10-30:
> You want to say it is cleared in browser not in esp-link cache?
Yes.

All new information transmitted to clients and not recorded to server.
All client clear request are not transmitted to server side.
Some time information are recorded to the server and it's not possible to clear them.
All client side look good, all server side log are buggy.

tve, 2017-10-30:
What are you trying to accomplish? I.e., what is your use-case?

Serveurperso, 2017-10-30:
Look at https://www.serveurperso.com/?page=robot

This is my home autonomous robot controllable by anyone (if I give permission 1 clic for me).
It's a personal system for home (no lucrative) and soon an opensource autonomous robot the first autonomous working on a 80MHz class Arduino compatible micro controller !

The console log with server side cache is cool to help debug of new features.

Serveurperso, 2017-10-30:
https://www.youtube.com/watch?v=BMjCmAJvCgM
https://www.youtube.com/watch?v=8XHUm8PxUaI

tve, 2017-10-30:
Nice project, what are you trying to accomplish by clearing the log?

Serveurperso, 2017-10-30:
My accomplish is no bug on my system:) ESP-Link is very cool and I want a consistent web console.
I don't understand how work this random server side cache and why the client do not rely on server.
I want a server sync with the client / a consistent work like the two log on my page:)

Serveurperso, 2017-10-30:
- higher priority is the server side cache update, because I can miss a lot of usefull debug information when I debug my project, it's a major issue because a debug must never loss data.

- lower priority for the broken clear...

uzi18, 2018-01-02:
@Serveurperso - Pascal it's ok for You to miss some chars received beatween console refresh and "Clear Log" clicked?

Serveurperso, 2018-01-03:
Hello uzi18, It's not a problem. what is the polling frequency ?

uzi18, 2018-01-03:
Forget it, I have working solution,will provide patch soon.

Serveurperso, 2018-01-03:
Let me know when a binary is available for an OTA update (I use esp-link on two robot now)

uzi18, 2018-01-03:
I can send You my copy of binary but it is for 32Mb flash - is it ok for
You?

Serveurperso, 2018-01-03:
Cool I use 32MB esp8266ex

uzi18, 2018-01-03:
[esp-link-v3.2.47.alpha.4-g316c50b-dirty-Lucjan.tar.gz](https://github.com/jeelabs/esp-link/files/1601528/esp-link-v3.2.47.alpha.4-g316c50b-dirty-Lucjan.tar.gz)

Serveurperso, 2018-01-03:
My versions are updated from :
esp-link v3.2.47-g9c6530d
to :
esp-link v3.2.47.alpha.3-gb8605b5-dirty-Lucjan
Tested : succes

1) TX data
2) verify the console -> data are OK
3) refresh the page
4) verify the console -> data are OK
5) clear data
6) refresh the page -> data are cleared OK
7) few other vicious combination OK

Now it look realtime sync to the server. it's closed for me Thanks a lot

Serveurperso, 2018-01-04:
What is the (server) buffer size ? because I can't clear when the client is downloading ! very slow for very large buffer.

Serveurperso, 2018-01-04:
Please ignore the (deleted) previous post

uzi18, 2018-01-04:
1024 chars

Serveurperso, 2018-01-04:
Sorry this is a mistake please ignore this post

commit 9d5bfffc (2019-08-02): - add ajax /console/clear - fix #348
closed 2019-08-02 by tve