jeelabs/esp-link #352: Increase max MQTT message size

Rotario, 2018-01-02:
Hi,
I have MQTT running fine over SLIP when the total serial buffer is less than 1024 including the SLIP overhead. I see that there's a hardcoded max slip buffer here:
https://github.com/jeelabs/esp-link/blob/e57f29c019b1d895d022f2a35e54201ff99df8cc/serial/slip.c#L31
Is there an easy way to automatically pack the MQTT message payload into multiple buffers? I think changing this max SLIP buffer may affect the general working of the system.

Thanks

uzi18, 2018-01-02:
> Is there an easy way to automatically pack the MQTT message payload into multiple buffers?

What do You mean exactly? Please provide code example for this idea.

Rotario, 2018-01-02:
So I can't send a large MQTT payload as I'm arbitrarily restricted by the max SLIP Buffer to 1024 bytes minus the overhead. Is there a way to circumvent the max SLIP data buffer without messing with that hardcoded line and rebuilding the ESP firmware.

For example with a 3000 byte MQTT payload publish by ELClient, the SLIP library splits it up into 3x1000 byte chunks and reassembles on the other end. I guess this would be nontrivial to implement?

uzi18, 2018-01-02:
Why do You need such long msg send to mqtt?
Sure, it is not trivial, as it was designed for max 1024 bytes.
Do You use mqtt example or different code base?

Rotario, 2018-01-02:
Thanks for your responses, I'm using the mqtt example.

In regards to the large message, I'm using a high frequency accelerometer running at 20kHz to store around 0.25s worth of values, then I want to dump these through MQTT as one message. Don't you think this is a valid application?

uzi18, 2018-01-02:
Do You realy need so many values, what for?
You can calculate avarage and send this every 0.25s.
Do You measured timeouts and overhead of SLIP/mqtt (less than 0,25s)?

Rotario, 2018-01-03:
I need the values so I can do a frequency analysis once the values are on the server. It probably takes less power to send the values as it does to perform an FFT on the micro itself. The timeouts and time overheads for sending the actual data doesn't really matter as I'm only interested in a 0.25s window, once this data has been sent I don't really need to send anymore, although it would be useful to keep a constant stream flowing. I don't think it's possible with the slip restriction.

uzi18, 2018-01-03:
Ok so how big buffer exactly You need?

Rotario, 2018-01-03:
Around 2500 values, at 12bit resolution, and 4 bytes per value, as the mqtt.publish only takes char arrays even though the value itself is something like "2056". It's ok I have already written the code to automatically split it up into 128 value blocks. I just think it'd be sweet to construct the buffer then spit it out as one, making use of the huge max MQTT message limit.

tve, 2018-01-04:
If you have 12 bit values you only need 2 bytes per value, mqtt is binary. You could pack tighter, but it gets more complicated.
Overall, rather than making larger messages you're better off making more smaller messages. Yes, that increases overhead a little but there are also other concerns, such as packet loss, which increases with packet size. The reason for the max size in esp-link is memory allocation and memory fragmentation. It's easy to run out of memory due to fragmentation and packet loss. You have to remember that until TCP gets an ACK it can't free the sending buffer...

Rotario, 2018-01-04:
Thanks for the support!
I'll look at reducing the size, at the moment I'm creating a long string then returning a pointer to it for the MQTT publish method, I guess I'm better creating a char array.

At the moment I'm wondering whether to make the design choice of sending an MQTT payload in the form of a JSON to allow easier manipulation on the other end, is it worth just sending the raw data instead and parsing it on the other end?

closed 2019-11-21 by Rotario