jeelabs/esp-link #353: esp8266 as AVR ISP

dontsovcmc, 2018-01-11:
Is it possible to add AVR ISP function to esp-link?

This is possible for esp8266 as i see here: https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266AVRISP

jandrassy, 2018-02-18:
esp-link is for serial connection. if you have SPI connected esp8266 use [a firmware for that](https://github.com/JiriBilek/WiFiSpiESP)

roger-, 2018-03-25:
+1, I was hoping to be able to do ISP programming with esp-link.

jandrassy, 2018-03-26:
@roger- , you have an esp8266 module connected to AVR Arduino over SPI interface?

roger-, 2018-03-26:
Well I misunderstood the esp-link description and thought it would allow me to flash an AVR over SPI. I did also want to use it for adding WiFi support as well. 

The Arduino sketch for flashing over SPI seems straightforward. Would it be easy to integrate it?

jandrassy, 2018-03-26:
use [this ](https://github.com/JiriBilek/WiFiSpiESP/tree/OTA_AVR_ISP_(optional))
it is what you want

roger-, 2018-03-27:
Great, thanks!