jeelabs/esp-link #355: Can add the ability to upload remotely through the server.

mtrucc, 2018-01-20:
This project is great!
Can add the ability to upload remotely through the server.
Is the server to send a command, esp8266 will be able to automatically download the server-side files and programming Arduino

uzi18, 2018-01-20:
I use for this simple bash script on server.

mtrucc, 2018-01-20:
@uzi18 No, I mean esp-link can automatically listen for commands from remote servers after networking.
For example, I have a server iot.server.com.
When esplink is able to connect to the Internet, it starts listening for commands from iot.server.com or sends the IP address to the server so that the server can execute the upload process

uzi18, 2018-01-20:
You can send this IP to server by using ELClient library.

mtrucc, 2018-01-20:
@uzi18 My contact email is mtrucc@gmail.com Can you contact me, I will only use docker to compile the file, but I will not modify the file, I would like to know how to add ELClient library to esplink

uzi18, 2018-01-20:
You add ELClient to Your Arduino project and use there, i assume esp-link is connected to Arduino.

mtrucc, 2018-01-20:
@uzi18 This is not good, I expect esp8266 to be more perfect

mtrucc, 2018-01-20:
@uzi18 I may say incomplete, I mean, expect esp8266 have such a function, so you do not have to write in the Arduino UNO related procedures. Also, esp8266.h will not be used because esplink is affected after using esp-link?

uzi18, 2018-01-20:
You need to think about esp-link like serial bridge or ethernet card for Arduino :)

uzi18, 2018-01-20:
With ELClinet lib, You can say to esp-link to connect to one server and send there something (eg. IP)
It is realy simple.

mtrucc, 2018-01-20:
Yes, you're right, then I should add ELClient to esplink. Not Arduino.

mtrucc, 2018-01-20:
wow, it's easy for you, but it's hard for me because I only use Arduino IDE to write some very simple programs. Interestingly, before I understood esplink, I turned out to be using the Arduino IDE flash esplink to convert the hex code into hexadecimal and off-line to burn the Arduino. This makes me sad

mtrucc, 2018-01-20:
@uzi18  Can you teach me? My email has been sent to you.

mtrucc, 2018-01-21:
I found that I can not upload programs using WAN IP
COM to LAN IP : ok
but com to wan ip : failed

mtrucc, 2020-11-20:
too long time later..

closed 2020-11-20 by mtrucc