jeelabs/esp-link #356: MQTT Password Character Limit Too Short

asobey, 2018-01-23:
The Adafruit_MQTT server creates standard AIO Key/Password of 32 characters. I am unable to use that specific MQTT server because of the 31 character limit. Here is the error message I receive:

Error: Value for mqtt-password too long (32 > 31 allowed)

uzi18, 2018-01-23:
Where did You get this Error - please attach here screenshot

asobey, 2018-01-23:
When I get home I will attach an image, but in the meantime:

It occurred under the 'REST/MQTT' tab, in the 'Password' Field at the bottom. The error message mentioned above appears at the top of the page.

When I reduce the password by one character to 31 the page allows it.

asobey, 2018-01-23:
[image]

tve, 2018-01-24:
@asobey does uzi18's version fix your issue?

asobey, 2018-01-27:
I apologize for not being able to check this.  I looked at the code and it appears that it should work, but I am unable to BUILD the .bin files. I will keep trying.

uzi18, 2018-01-27:
If You want I will attach here my compiled version

uzi18, 2018-01-27:
[firmware.zip](https://github.com/jeelabs/esp-link/files/1670496/firmware.zip)

vctrvlad, 2018-03-27:
Hi!
Nope, that ain't working...

I am now scratching the surface of this IoT thing, and I get the same error message:
[image]

I am using this release:
esp-link v3.2.47.alpha.3-gb8605b5-dirty-Lucjan

should I go earlier?

uzi18, 2018-03-27:
get my firmware.zip from post above ...

vctrvlad, 2018-03-27:
Hello uzi18, 

As I mentioned, I have tried it already.
I have flashed user1.bin in the zip provided by you, and same behavior...
Maybe I am doing something wrong?

[image]

Also, please tell me what tools do I need and how to compile this project, I am using a Win10 machine.
I also have Mingw installed.

Thanks for your prompt answer.

asobey, 2018-03-27:
I too had trouble with that Zip file, but I did not want to comment yet, as I have not giving the problem enough of a troubleshoot as it deserves.

vctrvlad, 2018-03-27:
I think the problem is in the following conditional statement:

[image]

in the file cgi.c

vctrvlad, 2018-03-27:
and maybe there is a hiden problem deeper in the httpd.c file in the function that returns the length of the parameter:

[image]

?

uzi18, 2018-03-27:
only problem was not clean rebuild and var must have 65 chars minimum, this firmware is ok:
[firmware.zip](https://github.com/jeelabs/esp-link/files/1853871/firmware.zip)

vctrvlad, 2018-03-28:
ok, my bad then, I will give it a try...

commit 84fd0a3a (2019-08-02): Change MQTT password to 64 chars, fixes #356
closed 2019-08-02 by tve