jeelabs/esp-link #357: Help Adding better uC flashing interface using arduino ide

frizzle101101, 2018-01-27:
So i have this working manualy with just the firmware the way it is and a program on my computer that maps the esp-ip:23 to a COM port which i can use the arduino ide to target. I say manually because as soon as you see upload in the ide you must press the reset button and hope it syncs up. Ive gotten it to work a bunch of times. So i just want to purpose to add some checks to check for the serial sequence (27,1,0,1,14,1,20) then resets the child uC for the next upload attempt, this is then the start of the sync sequence it is timed propperly.

uzi18, 2018-01-27:
Try with port 2323

frizzle101101, 2018-01-27:
No difference on port 2323 same operation i must restart it manually to get it in sync to start the upload

frizzle101101, 2018-01-27:
the auto reset on 2323 doesnt help with the timing it.

uzi18, 2018-01-27:
if you have constant connection reset will not occur.

frizzle101101, 2018-01-27:
Ok i got that to work but the restart is too early for sync

frizzle101101, 2018-01-27:
I just need help finding the place where I can edit the code