jeelabs/esp-link #358: please , if possible , fix esp8266 flash another esp8266

ca4ti, 2018-02-13:
Hi , i try flash esp8266 ( https://github.com/jeelabs/esp-link/blob/master/UC-FLASHING.md ) , but i have error :

Note : i use a version : esp-link v3.2.47-g9c6530d , with esp8266-12

-------------
C:\Python27\Scripts>esptool.py.exe --baud 115200 -p socket://192.168.4.1:2323 flash_id
esptool.py v2.2.1
Connecting........_____....._____....._____....._____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to Espressif device: Invalid head of packet ('\x08')

C:\Python27\Scripts>
-------------

I not try using linux , but if work with linux , i try , with yours information .

ca4ti, 2018-02-13:
sorry more one note : i use esptool default installed of python *(pip install esptool)

tve, 2018-02-21:
I have never been able to flash one esp8266 with another one. See https://github.com/jeelabs/esp-link/blob/master/UC-FLASHING.md#flashing-an-attached-esp8266