jeelabs/esp-link #359: Sketch webserver_led.ino not working?

yesnoj, 2018-02-16:
Hi all, i've a problem, once i've uploaded the webpage , in my case the SimpleLED.html , i don't know how to remove this from the ESP8266. I've also reflashed with the nodemcu flasher, but it remains there. Another thing, every time i try to upload a sketch, the relative for the webserver_led, so webserver_led.ino, i can't access anymore to the 192.168.4.1 page, i've to reflash with the tool.
How can i solve this two problems?
Thanks in advance!

uzi18, 2018-02-16:
go to Web Server link on the left side, and click upload without choosing any file.

yesnoj, 2018-02-17:
uzi18, thanks a lot!It's working!Well is not the most intuitive way but, it works :D
However, can you tell me why if i flash a .ino file, than i can't access anymore to the webpace 192.168.4.1?For sure the server is running, becouse i can see the ssid in the wifi list...
Thanks in advance!
PS, im using the sable version esp-link v3.0.14 as suggested in the download page...and this for the .ino https://github.com/jeelabs/el-client/tree/master/ELClient/examples/webserver_led

yesnoj, 2018-02-17:
I've flashed the old version, the esp-link v2.2.3 - 2016-06-21 21:58:48 - 1bcdc62, but the webserver pages are not available...if i try to reflash to the v3.0.14, it always load the 2.2.3!What i'm doing wrong?! :/

yesnoj, 2018-02-23:
Can someone help me to solve this issue?Maybe i'm doing something wrong?!
@tve @uzi18

yesnoj, 2018-02-28:
UPDATE: I was totally wrong with the project concept...i was flashing the .ino file in the same board (wemos d1 mini) where ESP-link was installed, and this is not correct. Thanks @ieee488 for our conversation.
I've to flash the .ino on another board, for example a Nano (i've now ordered a pro-mini), and link this board to the D1.
However, i can't understand why is not possible to flash .ino file in the same board...i've seen that i can select a partition (SPIFFS 3M,1M) in the ide. Can the esp-link be flashed in the 3M partition, and the .ino in the 1M?This will be possible in the future?!

uzi18, 2018-02-28:
Sorry I have missed your message,  esp-link splits flash into some parts depending on flash size addresses are different. 
It is necessary to not mess with user1 and user2 they are dedicated.
Also binary file prepared in arduino ide must be flashed from address 0.
When user1 starts from 0x1000.
If esp-link uses actualy user2 it will slash bin onto user1 so not from address 0 but 0x1000 thats why it is not working. 

So don't mix esp-link firmware with arduino generated firmware

yesnoj, 2018-02-28:
Don't worry @uzi18 , i've flashed correctly all the stuff following the guide and is working perfectly. Was my fault to flash the .ino file in the same D1 with esp-link already flashed :/
I've always flashed user1 with the flasher app.
@uzi18 are you on hangout too!?

uzi18, 2018-02-28:
Yes I'm,  you can find me also on gitter.im

yesnoj, 2018-02-28:
i don't have access from here to gitter.im ... :/ how can find you on hangout?!

uzi18, 2018-02-28:
I dont want to put my private email here...

uzi18, 2018-02-28:
Ok go here https://defuse.ca/b/sQEoUQCCb7dN7BWcGExc6M
And put pw: esp-link

yesnoj, 2018-02-28:
Please close this "not issue" ...thanks to users @uzi18  and @ieee488 for their clarifications :)