jeelabs/esp-link #360: Wemos D1 does not create Access Point

ivanplkolev, 2018-02-17:
Hello I am trying to instal esp-link, I try with 2 different boards - Wemos d1 mini and CH340G and the result is the samo, I try with NodeMCU Flashed and esptool.py, here are the logs from my esptool.py and Wemos D1, it seems to be uploaded, but when I search for networks from my phone I do not see : ESP_012ABC , What I am doing wrong ?

pi@raspberrypi:~/Documents/esptool/esp-link-v3.0.14-g963ffbb $ esptool.py erase_flash
esptool.py v2.2.1
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 1.9s
Hard resetting...
pi@raspberrypi:~/Documents/esptool/esp-link-v3.0.14-g963ffbb $ esptool.py --port /dev/ttyUSB0 --baud 230400 write_flash -fs 32m -ff 80m     0x00000 boot_v1.6.bin 0x1000 user1.bin     0x3FC000 esp_init_data_default.bin 0x3FE000 blank.bin
WARNING: Flash size arguments in megabits like '32m' are deprecated.
Please use the equivalent size '4MB'.
Megabit arguments may be removed in a future release.
esptool.py v2.2.1
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 230400
Changed.
Configuring flash size...
Flash params set to 0x004f
Compressed 3856 bytes to 2763...
Wrote 3856 bytes (2763 compressed) at 0x00000000 in 0.1 seconds (effective 235.9 kbit/s)...
Hash of data verified.
Compressed 330756 bytes to 244619...
Wrote 330756 bytes (244619 compressed) at 0x00001000 in 11.1 seconds (effective 237.8 kbit/s)...
Hash of data verified.
Compressed 128 bytes to 75...
Wrote 128 bytes (75 compressed) at 0x003fc000 in 0.0 seconds (effective 130.0 kbit/s)...
Hash of data verified.
Compressed 4096 bytes to 26...
Wrote 4096 bytes (26 compressed) at 0x003fe000 in 0.0 seconds (effective 5163.4 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting...
pi@raspberrypi:~/Documents/esptool/esp-link-v3.0.14-g963ffbb $ esptool.py read_mac
esptool.py v2.2.1
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Uploading stub...
Running stub...
Stub running...
MAC: 68:c6:3a:9f:70:4b
Hard resetting...
pi@raspberrypi:~/Documents/esptool/esp-link-v3.0.14-g963ffbb $ esptool.py chip_id
esptool.py v2.2.1
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Uploading stub...
Running stub...
Stub running...
Chip ID: 0x009f704b
Hard resetting...
pi@raspberrypi:~/Documents/esptool/esp-link-v3.0.14-g963ffbb $ esptool.py flash_id
esptool.py v2.2.1
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Uploading stub...
Running stub...
Stub running...
Manufacturer: c8
Device: 4016
Detected flash size: 4MB
Hard resetting...

tve, 2018-02-21:
Your best bet is to attach a serial port to gpio2 and look at the debug output when it boots.

Chrisoo, 2018-03-01:
https://d3higte790sj35.cloudfront.net/images/mj/vg/a2ee5d93dcac94cded93dee502fb1ede.png

closed 2018-03-21 by ivanplkolev

ivanplkolev, 2018-03-21:
The problem was that in the video was not paid attention to the 4 checked checkboxes of the 4 binary files in nodemcu  flasher.

uzi18, 2018-03-21:
Please give us screenshot here for future questions.