jeelabs/esp-link #361: result: Invalid/unknown record type: 0x03

ieee488, 2018-02-20:
I am able to successfully upload a small sketch to an Arduino Nano clone.

At the end of the upload, I see **AVR in sync** and then **result: Invalid/unknown record type: 0x03**

I can go back and forth between two sketches that print out different things to the Serial Monitor.
That's how I know the uploads are successful, but I don't know what that text means.

uzi18, 2018-02-21:
How to reproduce this issue?

ieee488, 2018-02-21:
avr-link flash --esp 198.162.0.104 --file feb20.hex -v

avr-link  0.5
Windows 10 64-bit
esp-link 3.0.14
ESP-01
Arduino Nano clone

uzi18, 2018-02-21:
Isn't it avr-link bug?

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Wolny
od wirusów. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

uzi18, 2018-02-21:
try avrflash script from esp-link

2018-02-21 3:01 GMT+01:00 Bartłomiej Zimoń <cactus.list@gmail.com>:

ieee488, 2018-02-21:
What do you mean avrflash script?

How do I do that?

uzi18, 2018-02-21:
https://github.com/jeelabs/esp-link/blob/master/avrflash

ieee488, 2018-02-21:
I don't know what that mean.

uzi18, 2018-02-21:
Try upload with script: avrflash

ieee488, 2018-02-21:
I am running on Windows 10.
Will avrflash work?  It looks like it is for Linux.

uzi18, 2018-02-21:
If You got RPi you can use it from there