jeelabs/esp-link #362: after off/on esp-link cannot be found

ieee488, 2018-02-21:
I flash esp-link 3.0.14 to ESP-01 module.
I configure it to join my network. It gets IP address 192.168.0.104
Everything is fine.

I remove power.
Wait a few minutes.
I apply power.
I go to 192.168.0.104  and I cannot connect.

What am I doing wrong?

yesyesuk, 2018-02-21:
Maybe it gets a different IP address? Without a fixed IP lease from your DHCP server you never know whether you will get the same IP address each time.

Try to configure the ESP with a fixed IP address outside the range of IP addresses that your DHCP server provides and see if you can access it each time then.

ieee488, 2018-02-21:
Every time I download esp-link firmware again, and go to 192.168.4.1 to configure the wifi, it always gets 192.168.0.104 from my router.

It is really getting frustrating. 

I don't know if I should try 2.2.3 but https://github.com/jeelabs/esp-link/releases/download/v2.2.3/esp-link-v2.2.3.tgz does not have esp_init_data_default.bin that is in 3.0.14

I have no idea what is going on with esp-link
Documentation isn't that clear for newbie like me.

People write that they have esp-link working, but I have no idea which version they used and if it is broken now.

uzi18, 2018-02-21:
If you got such situation try push reset button on module. It work?

ESP need lots of power on start - if You add 2000uF capacitor between +3V and GND on module it will work better.

ieee488, 2018-02-21:
I have the ESP-01 so there isn't a reset button on the module.

I have a dedicated 3.3V AC adapter for the ESP-01. 
I don't have a 2000uF capacitor.
I think 100uF is the largest I have.

tve, 2018-02-21:
The best you can do is to hook up a serial port to uart2/gpio2 and look at the debug messages. 115200 baud or 76600 (approx) baud. That will tell you what is happening.

ieee488, 2018-02-21:
After the firmware is downloaded, am I suppose to set the module for AP+STA mode?

The documentation is not clear.

I just want to reliably connect when I power down then power up, and the module with esp-link 3.0.14 **does not power up reliably*

.

tve, 2018-02-21:
The wifi settings are not programmed when you flash esp-link itself. You can flash the blank.bin over the flash page that holds the wifi info to reset the settings.

ieee488, 2018-02-21:
I re-flash everything including the blank.bin
That re-flash works fine.

It is only after I power down the module, that I cannot connect to it when I power it up again.

It is dead.

Nothing on GPIO2.

ieee488, 2018-02-21:
ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 2408, room 16 
tail 8
chksum 0xe5
load 0x3ffe8000, len 776, room 0 
tail 8
chksum 0x84
load 0x3ffe8310, len 632, room 0 
tail 8
chksum 0xd8
csum 0xd8

2nd boot version : 1.6
  SPI Speed      : 40MHz
  SPI Mode       : QIO
  SPI Flash Size & Map: 8Mbit(512KB+512KB)
jump to run user1 @ 1000

¡ á[00][00][00][00][00][00][00][00][00][00][00][00][00][00]

uzi18, 2018-02-21:
@tve isn't it like unresolved issue with firmware 3.2.x?

ieee488, 2018-02-21:
But I am using 3.0.14   or at least that is what it is saying

ieee488, 2018-02-21:
If I can use 2.2.3, what do I do about the missing esp_init_data_default.bin  ?

ieee488, 2018-02-25:
So, there is no answer?

I have two ESP-01 modules programmed with 3.0.14 of esp-link which cannot be found when they are powered down then powered on.

Modules work fine until they lose power. I can configure them. Everything.

Does no one think of testing this problem?

Do you expect the module to never lose power?

This is so fundamental, but there is silence on this problem.

uzi18, 2018-02-25:
Do You have CH_PD pin connected to +3V?

ieee488, 2018-02-26:
https://iotdiary.blogspot.sg/2017/09/using-esp-link-transparent-bridge.html

except I have the Nano and not the Pro Mini

He claims he uses 3.0.14 and it works for him.
But I don't know if he ever powered down then powered up.

Every time without fail, when I first power up the ESP-01, I can never connect to it.

If I am lucky, after an hour, "maybe" I can connect.

Or if I  do the power off/power on a few times, then I can connect.

**## I never can connect on first try.**

Maybe there is a setting on my router that needs changing, but so far, esp-link does not work consistently for me. **And no one here has bothered to try this and verify this power down/power up issue**. It is so simple, but people here cannot be bothered to try it.

yesyesuk, 2018-02-26:
That's a bit of a harsh tone that does not encourage people to help you. Remember, this is a project that someone does in their spare time and makes it available to everyone for free. This isn't a commercial product with service level agreements!

Besides, it works fine for everyone else. I power mine up and down often and never had this issue.

ieee488, 2018-02-26:
I am frustrated because the problem only happens with power cycling.

What am I doing wrong?

Is it a router setting? WHAT?

It doesn't work consistently for me, and that is the reality.
I am not going to lie and say it works when it doesn't.

And so far, I have not received any helpful things to try.

yesyesuk, 2018-02-26:
A good start might be answering the questions you were asked here. Most importantly, how did you wire up the ESP? Do you have CH_PD pin pulled up to 3.3V through a resistor? Do you have GPIO0 pulled up to 3.3V when not flashing?

ieee488, 2018-02-26:
Did you even bother seeing the link I linked?????  18 minutes ago

tve, 2018-02-26:
not sure I even want to chime in given the tone here... I've rarely had a problem with reboot or power cycling. I certainly do it all the time. One thing to note is that when you flash the chip the programmer (e.g. esptool) can issue a "run" command, which bypasses the usual boot selection that is built into the hardware that uses gpio0, gpio2, and gpio15 to decide where to boot from. For example, if gpio15 is pulled up the just-loaded program will start right after programming but will never start again after a HW reset. This may or may not be your issue...

gheesung, 2018-02-27:
@ieee488 I am the author of the blog post you have mentioned. 

Firstly, I like to take this opportunity to thank @tve for providing this excellent firmware and el-client library, I have created a few mini projects using the same firmware. Recently, I have built a touch panel controller using a Mega2560 and ESP8266 board to control some Sonoff smart switches using ESP-Link. I will find some time to do a write-up.

For the mini project mentioned in the blog post, I turn on occasionally to do some testing but the ESP-link setting will never disappear. Initially, I have similar problems as mentioned when I try to power both the ESP and Pro-mini using the serial pin header. Later, I change it to supply the power to the RAW pin using a 5V and 1A supply (most portable chargers has this) and everything is working fine.

In a short technical blog post, I am trying to demonstrate a concept with an assumption that the reader has the prior background. I certainly welcome feedback so that I can improve on it.

uzi18, 2018-02-27:
@ghesesung it true, esp need lot of peak power. It is necessary to add big capacitor sometimes very near esp to buffer energy for such peak.
2000-2800uF should be ok.

gheesung, 2018-02-27:
For my case, the ESP-01 is powered by the Pro Mini regulated 3.3V output and I think it is already a stable voltage source so I didn't add any cap.

uzi18, 2018-02-27:
@ieee488 did You even tried to test another esp module?

ieee488, 2018-02-27:
As I wrote before, I have a dedicated 3.3V AC adapter. Not using Arduino Uno's 3.3 V.

And as I wrote before I have two(2)  ESP-01 modules that flashed with esp-link.
These were purchased several years ago 2016 not 2017.
Both do the same thing.

I have two other ESP-01 modules still untouched by esp-link. And I will not do anything to them with esp-link.

uzi18, 2018-02-27:
Did You checked esp-link 3.2?

ieee488, 2018-02-27:
No, I haven't tried 3.2
I am going to leave this project for awhile and come back to it with fresh eyes.

uzi18, 2018-02-27:
Just upgrade firmware and we will know more about problem, try to add
capacitor for 3V on module,
what is flash chip (photo/ID), we need more information to find solution
for You.

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Wolny
od wirusów. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

uzi18, 2018-02-27:
https://github.com/jeelabs/esp-link/releases/tag/v3.2.47.alpha

2018-02-27 21:20 GMT+01:00 Bartłomiej Zimoń <cactus.list@gmail.com>:

uzi18, 2018-03-02:
@ieee488 please show me esptool cmd to flash esp-link firmware

ieee488, 2018-03-02:
I am on Windows 10.
I used Espressif's Firmware Download Tool 3.6.3

coldpenguin, 2018-03-07:
So I have been using the latest head (well last weekend) compiled on windows with the 1.7 bootloader. I noticed that the flash command wasn't suitable for my 'esp-01s' and was having similar issues as related here. Might try and tidy up and submit a pull request if possible at the weekend.
However, the log output above looks ok to the extent it goes, could do with reproducing it at 115200 baud also as I suspect that was a working scenario.
Having a diagram/wire up of what the op actually has done would help, as has already been stated by the op they have diverged from the 'tutorial' already.
Symptoms I saw were that on second boot the module seemed to revert to users, which is not flashed in the example.
On an ota update the rI'm did swap back correctly, and as far as I could be bothered to investigate stayed correct (sorry am comparing twould different systems to see which would better meet my needs, so once working wasn't too concerned to debug further)

ieee488, 2018-03-07:
**## Already posted above what the "wire up" is.**

[https://iotdiary.blogspot.sg/2017/09/using-esp-link-transparent-bridge.html](url)
the author of that "tutorial" claims it works for him

uzi18, 2018-03-07:
Try to flash exactly like author, with esptool

ieee488, 2018-03-07:
I don't have esptool. I am on Windows 10 laptop.
I don't want to keep repeating myself.

uzi18, 2018-03-07:
It is hard to help You man, You can use esptool even on windows 10.
What I'm thinking of is to reproduce exactly same "environment" like author.

ieee488, 2018-03-07:
I will have to work on this when I have time.

This seems to grasping at straw time when the actual firmware is the same but the tool is different.

uzi18, 2018-03-07:
On some modules it is must to force DOUT as flash communication method.

ieee488, 2018-03-07:
Are you saying that the Espress Flash Download Tool doesn't do this? ? ?

uzi18, 2018-03-07:
I'm not sure, and You are?

uzi18, 2018-03-07:
Sorry I have no time if You don't give us any more clue.

ieee488, 2018-03-07:
I am not sure what you mean by **must force DOUT as flash communication method**.

I find it very strange that Espressif would not know how to flash it's own ICs.

swilson86, 2018-03-07:
Try this. If not working you can try selecting DIO or DOUT (Under SPI mode). Some of the chips made in late 2017 are fussy on this setting. I also assume you are using an ESP-01S chip. The S version has a 1 Megabyte (8 Megabit) FLASH size that will work with Over The Air flashing.

[image]

ieee488, 2018-03-08:
Thanks.

Yes, I have the ESP-01S with the 1 Megabyte (8 Megabit) flash.

That is how I have the ESP8266 Download Tool v.3.6.3 set up except that because I have a ESP-01, the addresses are 0x0FE000 instead of 0x3FE000 and 0xFC000 instead of 0x3FC000.

.
.
.

coldpenguin, 2018-03-08:
I still maintain you haven't posted your wireup, you have posted someone elses.
I have seen a lot of people make 'optimisations' which just don't work. Like using different chips.

I am on a Windows 10 laptop, I use esptool, I found it to be more tolerant than the espressif tool. The DIO option can be seen in the screenshot above just to the left of the middle. Can you provide the 74kbaud debug from the serial at the second boot?
The espressif tool I think is capable of jumping direct to executable code without a reset. Therefore if the bootloader is borked or has the wrong settings it can be bypassed. (I had an issue with tryING to use gpio 9 and 10 on a qio chip because of the bootloader reboot stuff).

Set the flash to 26m flash size to 1m
Flash mode to DIO.

Oxfb000 blank
0xfc000 esp unit default
0xfe000 blank
0x00000 boot 1.7
0x01000 user1
0x81000 user2

ieee488, 2018-03-08:
Using different chips in the wireup????
Wtf!
It is just the module!

ieee488, 2018-03-08:
The 3.3V is an AC adapter I bought. 

The Arduino is not in the picture.

I am simply trying to get to 192.168.102 which is the static IP of the ESP-01 after I power cycle it.
Why is that so difficult for people to understand????

The ESP-01 is unreachable from a PC on my home network. The PC is connected to the router which is at 192.168.0.1

Is that simple enough for people to understand?

uzi18, 2018-03-08:
3V3 and power 10W, what is a type model of this AC adapter?
Give us some real pictures, explain/try all we asking for before.
Difficult as You don't listen what we need, please cooperate.

ieee488, 2018-03-08:
It is a no-name brand I bought on eBay.
3.3 V  1A

coldpenguin, 2018-03-09:
I am thinking the debug logs at 115k might actually be more help now.

How is the network configured? At the start of the post you mention that the config is from the router (by that I take it dhcp) with one ip address. Later you say it is static with a different ip address. Also you ask questions about sta+ap mode.
If any of the netmask/broadcast is incorrect for the device and he network it might be connecting and not be able to communicate (with the caveat that if it is the broadcast address, communication might be possible until the arp cache expires locally, which could explain why it appears to work until a reboot). Additionally if you are in ap mode as well, if the channel for the ap doesn't match that of your sta then you might not be able to connect to your router. This is of course not an issue at the first configuration of connecting to your router after doing a network scan.

It might be worthwhile completely disconnecting the arduino from the system. Yeah ok, this means that it isn't working in the config you want it to, but it isn't working now either. It will remove/reduce the possibility of slow power up or groundin issues. Instead confirm your network config, router channel number and add a resistor from gpio 0 to 3.3v
A cap as close as you can get it to the esp wouldn't go amiss. Anything is better than nothing. It could be something as stupid as when you have the esp configured it boots to full power attach to your router rather than an an access point and just tips the power budget over the edge during that initial phase.

uzi18, 2018-03-09:
@coldpenguin, I think we try too hard to help and receive always almost no
answare for our questions.
If answare then only on one question.
Always it is our foult that we don't understand.

For me it is good idea to start from beginning so minimal connections
needed for esp, with additional CAP and usb serial converter connected to
sniff what is on debug port at 115200.
This is start condition, if everything is ok so add next element and repeat
test.

yesyesuk, 2018-03-09:
I have given up helping here a while ago. This guy has clearly never heard of "help us to help you", is too lazy to draw a diagram of his wiring, and has an attitude.

ieee488, 2018-03-09:
**EXACTLY** as https://github.com/jeelabs/esp-link/issues/url !!!  except I don't even have Arduino.

Just want to talk to ESP-01 !!!

Sorry that is not clear enough for you.

ieee488, 2018-03-09:
@coldpenguin 

TP-Link router 192.168.0.1   (default not changed)
reserve  192.168.0.102
reserve  192.168.0.105

ESP-01  static IP  192.168.0.102, gateway 192.168.0.1, net mask 255.255.255.0
ESP-01  static IP  192.168.0.105, gateway 192.168.0.1, netmask  255.255.255.0

uzi18, 2018-03-10:
You want to connect esp to esp by using esp-link?

ieee488, 2018-03-10:
**computer to esp-link's webpage**  

You know the one that is flashed into the ESP-01 
You know the one you show on this project's home page.
You know the one that is yellow and red.

[https://cloud.githubusercontent.com/assets/39480/19333951/73fcdcbe-90ad-11e6-8572-5e654377275a.png](url)

uzi18, 2018-03-10:
What about Debug log after flashing with esptool in DOUT mode?

ieee488, 2018-03-10:
I don't have a print out.
I can go to all the pages. I can scan for access points.

It shows that it is at 192.168.0.102 or 192.168.0.105 etc

What am I suppose to be looking for ? ? ?

swilson86, 2018-03-10:
uzi18, you have unlimited patience. ieee488, here it is in your language: On the page that is red, yellow, look on the left for "Debug log", cut and paste all the giberish in the middle of the page and post it here. Do this right after you flash the little square chip marked ESP-01S and make sure u flash it with the DOUT mode thingy ticked. GOT IT? Do it and stop wasting everyones time.

ieee488, 2018-03-13:
You guys should standardize on which flash mode to use.
You say DOUT, but https://github.com/jeelabs/esp-link/blob/master/FLASHING.md#initial-serial-flashing  says QIO.

Hmmmmm. inconsistencies

[27 more replies]