jeelabs/esp-link #363: Wifi not starting on Uno Wifi Dev

andrewmartin00, 2018-03-04:
Hi,

I have a Uno Wifi Dev board and I've been struggling a lot with getting it working.

I think that this FW will probably be the most useful way for me to use my board - and the forum suggests it should be compatible.

However when I flash the FW - the board isn't bringing up the wifi at all.  The Wifi Loght on the board isn't illuminating at all.

Here's what I've tried so far

* Successfully flashed V3.0.14 - no wifi hotspot
* Successfully flashed V2.2.3 - no hotspot
* Erased the flash on the 8266 - in case there was old configuration from the old FW causing issues
* repeated the installation of both V3 and V2 with no luck

Here are the commands I used

esptool-windows.exe --port COM4 --baud 9600 write_flash --flash_size 32m --flash_freq 80m --flash_mode qio 0x00000 boot_v1.5.bin 0x1000 user1.bin 0x3FE000 blank.bin

esptool-windows.exe --port COM4 --baud 9600 write_flash --flash_size 32m --flash_freq 80m --flash_mode qio 0x00000 boot_v1.6.bin 0x1000 user1.bin 0x3FC000 esp_init_data_default.bin 0x3FE000 blank.bin

As far as I can tell from other forums (e.g. this one https://github.com/arduino-org/arduino-firmware-wifilink/issues/23) the flash_size is correct.  

Oh - and I am loading the ESPRecovery sketch and pressing the button to put it into programming mode before starting the esptool command (when I don't do that the esptool command fails to work)

Can anyone who has done this help?  I can't even work out how to get any debug off the board - but it feels like the 8266 isn't booting at all

uzi18, 2018-06-09:
if You want i have full backup with latest working git - ap mode turned on.

jandrassy, 2018-06-13:
did you use the 40 MHz setting?