jeelabs/esp-link #366: ESP-01s flash ok but not boot AP

matiasojedam, 2018-03-24:
Hi

I flash my ESP-01s

~/esp-link-v3.0.14-g963ffbb# esptool -v -cp /dev/ttyUSB0 -cb 460800 -cd none -bz 1M -ca 0x00000 -cf boot_v1.6.bin -ca 0x01000 -cf user1.bin -ca 0xFC000 -cf esp_init_data_default.bin -ca 0xFE000 -cf blank.bin
esptool v0.4.6 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
setting flash size from 512K to 1M
opening port /dev/ttyUSB0 at 460800
opening bootloader
resetting board
trying to connect
trying to connect
Uploading 3856 bytes from boot_v1.6.bin to flash at 0x00000000
....
Uploading 330756 bytes from user1.bin to flash at 0x00001000
....................................................................................................................................................................................................................................................................................................................................
Uploading 128 bytes from esp_init_data_default.bin to flash at 0x000FC000
.
Uploading 4096 bytes from blank.bin to flash at 0x000FE000
....
starting app without reboot
closing bootloader

but when I boot the ESP, this keep blue light flashing fast all time without any response (no boot AP)

Any idea?

Thanks
Im using this power up wiring:
[image]

leodesigner, 2018-06-03:
Try to set port speed to 78400 and look to the boot loader debug messages.

uzi18, 2018-06-03:
And 115200 in debug messages on serial port

yossioo, 2020-06-02:
Came here as well
Tried flashing `esp-link-v3.0.14-g963ffbb` and `esp-link-v3.2.47-g9c6530d` with esptool.py on NodeMCU V3 according to the [instructions](https://github.com/jeelabs/esp-link/blob/master/FLASHING.md#initial-serial-flashing).
 Flash is successful, but no AP

Listening to serial on 78400 yields:
```
[17:10:05:719] load 0x40100000, len 2408, room 16 ␍␊
[17:10:05:727] tail 8␍␊
[17:10:05:727] chksum 0xef␍␊
[17:10:05:727] load 0x00000000, len 0, room 0 ␍␊
[17:10:05:736] tail 0␍␊
[17:10:05:736] chksum 0xef␍␊
[17:10:05:736] load 0x00000000, len 0, room 8 ␍␊
[17:10:05:744] tail 0␍␊
[17:10:05:744] chksum 0xef␍␊
[17:10:05:744] csum 0xef␍␊
[17:10:05:744] csum err␍␊
[17:10:05:744] ets_main.c 
```

Is there something I am doing wrong?
Or is there a way to fix this?
Thank you

asmagill, 2020-06-02:
I recently got a batch of ESP-01s's that wouldn't seem to work -- flashing looked like it worked, but the device kept resetting during boot (attach to the serial interface at 74880 baud to see the initial debug info the ESP outputs during a reboot).

I was able to finally get all of them to work by adding `-fm dout` to the esptool args (before the address and file mappings -- in my case, I used: `esptool.py --port /dev/tty.usbserial-A50285BI  --baud 460800 write_flash -fs 8m -ff 40m -fm dout 0x00000 boot_v1.6.bin 0x1000 user1.bin  0xfc000 esp_init_data_default.bin 0xfe000 blank.bin`

The possible options for `-fm` are `qio`, `qout`, `dio`, and `dout`... I don't know a way to determine which is required, I just tried them all until one of them worked.

Hope this helps!

yossioo, 2020-06-03:
Yep, flashing with `esptool.py --port /dev/ttyUSB0 --baud 230400 write_flash -fs 32m -ff 80m  -fm dout     0x00000 boot_v1.6.bin 0x1000 user1.bin     0x3FC000 esp_init_data_default.bin 0x3FE000 blank.bin` solved it!