jeelabs/esp-link #367: Is there any method to ensure security in Flashing avr remotely?

szjiajin, 2018-03-26:
I have spent some time learning this module and I have to say it is very useful! Thank you for all the hard work. I have two questions in my mind right now.

1. To my understanding any machines in the same LAN with esp8266 board could send the flash command therefore it is not very safe. Is there any method to do a verifying process before flashing MCU attached?
2. Although esp8266 has a LAN IP, it does not have a WAN IP so server on other network cannot send the flash command to esp board. Is there any way to flash attached MCU via Internet?

Thank you in advance!

susanin-ivan, 2018-04-10:
Hi,

@szjiajin , regarding your #2 - It is posible by setting correct routes on all involved gateways. 

For example, my setup:

ESP12 ---> BlackSwift Board (OpenWrt) ---> Room's wifi ---> Main GW ---> Internet

ESP12: 10.10.20.55 from BlackSwift (std dhcp alloc)
BlackSwift: dhcp 10.10.20.0/24 to internal wifi clients, external 172.21.1.20 from Room's wifi (controlled dhcp allocation)
Room's wifi: dhcp 172.21.1.0/24 to internal clients, external 10.10.10.123 from Main GW 
My PC:  10.10.10.222 from Main GW (std dhcp alloc)

To reach ESP12 (10.10.20.55) from my PC (10.10.10.222) I do once:

My PC # route add 10.10.20.0/24 gw 10.10.10.123
Room's wifi # route add 10.10.20.0/24 gw 172.21.1.20

Now My PC know where to route packets for ESP12 and involved Room's wifi also know where to find ESP12's network

By adding same route as on on My PC to Main GW will allow to reach ESP12 from Internet (well, not so good idea because of yours #1, but ssh tunneling can help here)

Hope this helps :)

Best regards,
Ivan

rtek1000, 2018-08-03:
Hello,

A way I tested: change the reset pin, this can be done in the configuration page of the esp-link, in Pin assignment.

The default for the ESP-01 is in GPIO2, but can be disabled, or changed to another pin as GPIO0.

Another way would be for the avr to intercept or block access to the reset, this needs a specific hardware scheme, but it might allow for password deployment, avr could receive a password via serial port to enable access to the reset.

Without access to the reset pin, avrdude informs the wrong parameter and the programmer does not respond.

On the internet, the cool thing would be to be able to read email and download the attached update file. I do not know if the ESP8266 has this capability.

overisberg, 2018-08-03:
Have a look at my "Security Pin" idea:
https://github.com/jeelabs/esp-link/issues/19