jeelabs/esp-link #368: Empty uC console, how to debug it?

gaggio, 2018-04-06:
Hi,
I successfully flashed v3.0.14-g963ffbb on an Amica NodeMCU (4Mb), and the wifi connection went smooth, it all seemed to work correctly. I've connected it to a Opentherm Gateway (in case anybody here knows the board), and when I try to see the uC console I see absolutely nothing coming in. Am I at least supposed to see the echo of the characters I send? By the way, if I connect via telnet on port 23, I can see what I type but absolutely no response.
The Opentherm Gateway is fitted with LEDs which are flashing as expected, therefore indicating that the firmware is running correctly, so I believe it is a communication issue through the esp8266 serial. I tried the different presets under "pin assignment", maybe I should try again rebooting every time I change.
Any suggestion on how can I track the issue down?

fliphess, 2018-07-21:
which version of the otgw did you use? 

If you use the pcb from nodo-shop, you should set the jumpers to use the esp communication pins rather than using the ftdi pins. 
See the pdf manual for more information: https://www.nodo-shop.nl/nl/index.php?controller=attachment&id_attachment=47

gaggio, 2018-07-31:
thanks, that was actually the problem!
I couldn't find an english manual, but now that you pointed me in the right direction I could understand just from the pictures.

Thank you