jeelabs/esp-link #369: [Question] RobotDYN UNO Wifi

DJManas, 2018-04-07:
Hello,

this is more likely a question, than issue.

I have been able to flash ESP on [this board](https://www.aliexpress.com/store/product/UNO-WiFi-R3-ATmega328P-ESP8266-32Mb-memory-USB-TTL-CH340G-Compatible-for-Arduino-Uno-NodeMCU-WeMos/1950989_32806748844.html) with ESP-Link. It works correctly I can configure the ESP, but I want to use it for OTA for Arduino.

I had switched the dip switch to correct position PIN 1,2 ON, others OFF.

I had tried to launch command:
`~/.platformio/packages/tool-avrdude/avrdude -DV -pm328p -Pnet:192.168.100.40:23 -cstk500v1 -b115200 -U flash:w:firmware.hex:i -C ~/.platformio/packages/tool-avrdude/avrdude.conf`

But it returns:
```
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00
```

I had tried to „hijack“ this [thread](https://github.com/jeelabs/esp-link/issues/319). But then I had found [this blog](https://iotdiary.blogspot.cz/2017/11/esp-link-on-atmega2560esp8266-board.html). Where it is written:
**In the ESP-Link home page, the pin assignment is not used. Only the ESP TX0/RX0 is connected to Mega 2560. If you want to use the reset pin,  you have to connect the ESP GPIO to Mega 2560 Reset pin manually.**

But I have tried almost every ESP gpio pin connect to reset pin on UNO, without any luck.

Can you guide me, what am I doing wrong?

Thanks,
Regards,
Petr Sourek

gheesung, 2018-04-09:
Hi Petr,

I am the author of the blog you have mentioned. I do not own an Uno-ESP as mentioned in your question but a Mega-ESP board. I did not manage to get OTA to work on this board. On the other hand, OTA can work on ESP01 and Pro Mini configuration.

Regards
Ghee Sung

DJManas, 2018-04-09:
Hello,

thanks for your answer. I just found out this board home and foud out ESP-Link project which is totaly awesome and wanted to combine those 2 things together to get my job done. As far as I can remember statement from RobotDYN, the Mega and Uno should be the same thing (except that Mega is using different serial).

I have a bought a lot of ESP-01 for simpler sensors so i can use one of these for this thing, but lately I had discovered diferent board (ESP-WROOM32) which is far more better for my projects, thanks to 18 channel ADC, I will try to play with this.

Thank you again, when I will have a little time I will try to play with it a little more.
Regards,
Petr Sourek

gheesung, 2018-04-09:
BTW, do you know how to connect an external antenna to this board? it has a hole for external antenna but do I need to unsolder any "0" ohm resistor as in other ESP07 board?

DJManas, 2018-04-09:
Sorry but no, never added external antenna to ESP boards, have not needed it yet :-(

DJManas, 2018-04-13:
I was in communication with RobotDYN, I had layed the question (hope correctly).

> I had seen, that there is possibility to connect external antenna for ESP8266. Do I need to unsolder any „0“ ohm resistor as in other ESP-07 boards?

And their answer is here, hope it would be helpful.

> About external antenna.
> Need to remove this is L2 induction.

[image]

gheesung, 2018-04-14:
Hi Petr,

Thanks for the useful information. I will try out and let you know if it works.

Regards

DJManas, 2018-04-27:
You are welcome and thanks for testing.

jandrassy, 2018-06-13:
with Mega there is a problem with Mega. the build in flashing is for 328p. Mega goes only over telnet. and do not forget, you cant flash over Serial3. 

for Uno +WiFi a jumper from Uno reset pin to io 0 on esp8266 header should do the reset. for the esp8266 header see the schema on Robotdyn site