jeelabs/esp-link #370: Scanning for networks indefinitely (similar to #218)

susanin-ivan, 2018-04-10:
Hi,

Thanks a lot for great and usefull software!

I've got similar issue (#218 ), but slightly different - debug do show found networks, but still none on web page:

...
228> Flash map 4MB:512/512, manuf 0xE0 chip 0x4016
233> ** esp-link v3.2.47-g9c6530d: ready, heap=18960
...
275432> HTTP GET /wifi/scan: 200, 7ms, h=10760
276511> GET scan: cgiData=0 noAps=34
276665> scandone
276666> Scan done: found 33 APs
276666> bss1: DIRECT-76-HP DeskJet 4670 series �� (-79) <--- this one seems to be issue
276666> bss2: Ora (-85)
276668> bss3: HOTBOX-3A76 (-94)
276671> bss4: HOTBOX-709A (-90)

Firefox's log:

JSON parse error: SyntaxError: JSON.parse: bad control character in string literal at line 2 column 45 of the JSON data. In: {"result": {"inProgress": "0", "APs": [ {"essid": "DIRECT-76-HP DeskJet 4670 series��", "rssi": -80, "enc": "3"},

Again, thank you!

Si03, 2018-04-10:
I had the same issue (HP printer too), it's related to the ssid length. The ssid is exactly 32 characters long, which is fine, the code reserves 32 character spaces to store it, but is not allowing for a terminating null character which causes a buffer overflow somewhere (sorry I cant remember where)

For interested parties, affected lines:
[char ssid[32];](https://github.com/jeelabs/esp-link/blob/fe4f565fe83f05e402cc8d8ca3ceefbc39692a1f/esp-link/cgiwifi.c#L156)
[strncpy ..... bss_link->ssid, 32);](https://github.com/jeelabs/esp-link/blob/fe4f565fe83f05e402cc8d8ca3ceefbc39692a1f/esp-link/cgiwifi.c#L214)
I fixed this by declaring ssid[33] and doing a 
cgiWifiAps.apData[n]->ssid[32] = 0;

Si.

uzi18, 2018-04-10:
Will try to check sources and provide firmware update here if You need it

closed 2021-06-30 by susanin-ivan

uzi18, 2021-06-30:
Will prepare fix for this

uzi18, 2021-06-30:
Do you need to generate bin for you?

susanin-ivan, 2021-06-30:
I think this was fixed already in newer releases.

uzi18, 2021-06-30:
No it is not fixed

uzi18, 2021-06-30:
here you are, please report after test
[esp-link-v3.2.47.alpha.24-g74e2043-dirty-Lucjan.tgz.zip](https://github.com/jeelabs/esp-link/files/6743021/esp-link-v3.2.47.alpha.24-g74e2043-dirty-Lucjan.tgz.zip)

uzi18, 2021-06-30:
@susanin-ivan please test it, remember it is double zipped file

susanin-ivan, 2021-07-01:
Ack, i'll be at that place on weekend, will check. Thanks again for your help!

susanin-ivan, 2021-07-06:
Hi again.
Unfortunatelly, there is no more problematic device at that place anymore :(

uzi18, 2021-07-07:
You can simulate this situation with ap name 32 long even with your cellphone as host in tethering mode

CRSREE, 2022-09-28:
> I had the same issue (HP printer too), it's related to the ssid length. The ssid is exactly 32 characters long, which is fine, the code reserves 32 character spaces to store it, but is not allowing [...]

CRSREE, 2022-09-28:
Hi Si,

I have the same issue  related to the SSID length. Thank you so much for the binary files , this really solves the problem.

Actually , I want to modify the esp-link source code for my application. So it would be helpful if you can provide the source code  for this bug fix.

Thank you
CRSREE

CRSREE, 2022-09-28:
> You can simulate this situation with ap name 32 long even with your cellphone as host in tethering mode

> here you are, please report after test [esp-link-v3.2.47.alpha.24-g74e2043-dirty-Lucjan.tgz.zip](https://github.com/jeelabs/esp-link/files/6743021/esp-link-v3.2.47.alpha.24-g74e2043-dirty-Lucjan.tgz.zip)

Hello,

Thank you so much for the bin file generated , it works fine for me.
But actually I need to modify the source code for my application. So it would be great if you can share the new esp-link source code with this bug fixed.

Thank you,
CRSREE

uzi18, 2022-10-12:
@CRSREE is it other kind of issue in #554 ? Does my bin resolve also #554?

CRSREE, 2022-10-14:
Yes.. The same issue that the SSID length is more than 31 characters. I modified cgiwifi.c file in the esp-link as one of the comments from above.

#comment
 I had the same issue (HP printer too), it's related to the ssid length. The ssid is exactly 32 characters long, which is fine, the code reserves 32 character spaces to store it, but is not allowing for a terminating null character which causes a buffer overflow somewhere (sorry I cant remember where)

For interested parties, affected lines:
[char ssid[32];](https://github.com/jeelabs/esp-link/blob/fe4f565fe83f05e402cc8d8ca3ceefbc39692a1f/esp-link/cgiwifi.c#L156)
[strncpy ..... bss_link->ssid, 32);](https://github.com/jeelabs/esp-link/blob/fe4f565fe83f05e402cc8d8ca3ceefbc39692a1f/esp-link/cgiwifi.c#L214)
I fixed this by declaring ssid[33] and doing a
cgiWifiAps.apData[n]->ssid[32] = 0;

Si.

CRSREE