jeelabs/esp-link #379: Smart Home Control Panel using ESP-Link

gheesung, 2018-05-07:
I have put together a smart home control panel using ESP-link and EL-client using the ESP8266 + Mega2560 combi board.

The write-up can be found [here](https://iotdiary.blogspot.sg/2018/04/smart-home-control-panel-using-esp-link.html). 

I have 2 issues:
1. Sometime, the callback is not trigger when message is sent by the sender. I am very sure that the message is sent because the Openhab Habpanel and mqtt fx at my desktop has received the message. Is it EL-Client or ESP-Link problem? I have no effective way to debug this.
2. When disconnected from MQTT, the current reliable way is to perform a soft reset at Mega2560 so that ESP-Link and El-client can establish the connection. Is there another method without performing the soft reset?

uzi18, 2018-05-07:
Where to find code of this sketch to check/test it?

gheesung, 2018-05-07:
@uzi18 The source code can be found here.

https://github.com/gheesung/mega2560_smartpanel