jeelabs/esp-link #384: Character/line mode

KjellVerb, 2018-05-25:
Hi, in #97 I read that the default behaviour of the UART bridge is to transmit each character separately. However, when using the esp-link as a remote serial console to a processor, I don't get character mirroring or tab completion (which I do get with a wired serial connection).
Does this mean that the latest version of esp-link works in line mode or that something else is wrong on my side?

leodesigner, 2018-06-07:
After looking to the serbridge.c it should be byte by byte if they are coming with delay of one packet at least.

KjellVerb, 2018-06-07:
Sorry for forgetting to close this, it turns out that I needed to configure it in Putty (Config - Terminal - Local line editing - force off)

closed 2018-06-07 by KjellVerb