jeelabs/esp-link #386: STM32 flashing support.

leodesigner, 2018-06-03:
Hi,
I have made modifications to support STM32 based boards (blue pill) serial flashing.
Is the pull request will be welcome ?
Thank you.

leodesigner, 2018-06-04:
In case if anybody interested here is the my fork with STM32 support option.
https://github.com/leodesigner/esp-link/tree/master
You need to set 8E1 serial mode in console web page for flashing.
Than do something like this:
```
sudo socat -d -d pty,link=/dev/cu.remote,raw,echo=0,perm=0666 tcp:XXX.XXX.XXX.XXX:23
stm32flash -g 0x8000000 -b 230400 -w generic_stm32f103c.bin /dev/cu.remote
```

Flodee, 2018-10-23:
cool thats exactly what i am searching. I have an maple mini stm32f103 which can be programmed with the arduino environment + and esp8266-07s with your esp-link firmware now. I connected gpio5 to reset of my stm32 board and gpio4 to the boot0 pin. Do you know if there is a possibilty to easily flash from inside the arduino ide? you did not mention the reset and boot0 pin in your post. How do you have your stm32 connected? thanks in advance

leodesigner, 2018-11-26:
I don't remember exact config at the moment, the ISP pin connected to boot0 pin.
It is changing the boot mode fro STM32.
I am using Platformio with VSCode for development. It is much better than arduino IDE.

marciogranzotto, 2023-06-08:
I've implemented that with EspHome and https://github.com/oxan/esphome-stream-server
Just FYI, as @leodesigner said, you can change the serial mode to `8e1` or you can run stm32flash with `-m 8n1`